openrocket/openrocket #235: Improper Tube Cluster Mass Calculation

SRFirefox, 2015-08-26:
When creating a cluster using inner tubes, the base mass of the component is multiplied by the number of tubes in the cluster and that calculated mass is also multiplied by the number of tubes in the cluster.  The final result being that total_mass = component_mass x tube_count^2 rather than mass total_mass = component_mass x tube_count^2.

Overriding the component mass back to a single tube mass works as a band-aid for the problem.  Components internal to the clustered tubes such as motor blocks and parachutes do not have this issue.

teyrana, 2015-09-05:
saw this.  Working on it.  The problem lies somewhere in BasicMassCalculator.java and stages being children of stages. 
Trying to figure out how to refactor it now.

teyrana, 2015-11-16:
Fixed on the development branch.  Look for this in the next release!

kruland2607, 2015-11-16:
Dan, 

Did you issue a pull request for this change? I don't see the code. 

Kevin

teyrana, 2015-11-16:
No, it was caused by code in the unstable branch, so I fixed it there.

(And by unstable branch, I mean the repo on my account.  I could do a pull
request if you like, so that it's in the openrocket main repo, but I
certainly don't think the code is ready for release yet.)

teyrana, 2016-12-11:
This fix is now in the openrocket repo, 'unstable' branch.  It will be available in the next release.

Recommend we close this as resolved.

wcravens, 2017-10-28:
Dan confirmed that this is fixed in Unstable

closed 2017-10-28 by wcravens