openrocket/openrocket #236: Simulation time step display bug

lpherr, 2015-09-07:
Simulation time step dialog rounds display value without rounding internal value, so what is displayed to the user is not what is used for simulation. This occurs only after the cursor has left the time step text box (eg changing to different setup tab) For example:

[image]

Yet a simulation timestep of 0.005
[image]

dkingsley, 2017-06-22:
I have tracked this to the format string associated with the unitized conversion of the timestep to its string representation using the FixedPrecisionUnit class. The timestep units are setup to show two digits of  precision in UnitGroup.

If the user tries to set the timestep to 0.001 the conversion results in 0.00 and then when the simulation starts up with a timestep of 0.00 which is smaller that the default minimum timestep, the timestep is then is set to the minimum 0.001.

Also, moving the slider does not update the spinner but the spinner will update the slider.

hcraigmiller, 2022-08-24:
First, let me say that this is an area I know little about. But, attempting to duplicate the steps above, I cannot replicate the results shown in the above graph as to the simulation timestep (using A simple model rocket); the viewable data field is limited to two digits with a lower limit of .00.

> Also, moving the slider does not update the spinner but the spinner will update the slider.

In addition, the slider does update the spinner and the spinner does update the slider within the range of .00 to .20., inclusive.

It appears that this issue may have been fixed at some point since 2017. Should this issue be closed or clarified?

JoePfeiffer, 2022-08-25:
There still seem to be some vestiges of this left, but I can't see that it's important.

If I enter a timestep of .001, simulate, and plot (it had never occurred to me before that I could plot the simulation timestep.  That will be a useful debugging aid going forward), I get a plot showing that the timestep is indeed .001. If I go back to edit the simulation again, I find the timestep in the display window is .00, but when I run the simulation again the timestep is .001.

If someone wants to adjust the window to get wider for absurdly small timesteps, they're welcome to do it.

hcraigmiller, 2022-08-25:
@SiboVG, is increasing the window size by one digit a quick fix (limiting the input to 3 digits after the decimal)?

JoePfeiffer, 2022-08-25:
A timestep of anything under 1/100 of a second is absurd, and wanting extra decimal points beyond .01 (like someone wanting .015) is equally absurd.  There's no reason to extend by one digit; if we make any extension it should just grow as needed to accommodate the inevitable user who wants .0001.

hcraigmiller, 2022-08-25:
> A timestep of anything under 1/100 of a second is absurd, and wanting extra decimal points beyond .01 (like someone wanting .015) is equally absurd.

Should the timestep be capped at 2 digits (hundredths) so that a thousandth cannot be entered?

JoePfeiffer, 2022-08-25:
Thinking about it, that might not be a bad idea...

PR #1615 (2022-08-28): [#236] Limit simulation time step to 0.01 s
This PR fixes #236 and limits the simulation time step setting to 0.01 s.

Although this is a simple PR, I did implement something that **needs testing** of whether it doesn't brake other stuff in the program. I added a check to `setValue` in `DoubleModel.java` that clips the set value to the model's maximum and minimum value. I must say I was surprised that this wasn't implemented already. The effect of this operation should be that when you type a numerical value in a spinner box and hit enter, that if that value was larger than the DoubleModel's m [... cut]
closed 2022-08-28 by wolsen