openrocket/openrocket #246: Enhance our data collection from ThrustCurves

wcravens, 2015-09-30:
ThrustCurve actually has more data that we are currently using.  We could enhance our ThrustCurve import mechanism to include this extra data that exists outside of the eng/rse files

From kruland2607:
We download the set of .eng/.rse files from thrustcurve and run them through a standalone preprocessor (SerializeMotors.java) to produce a blob.  This blob is a Serialized representation of the internal datastructure used by the app.  This way it's stinking fast at start up.  A number of years ago, before I made this change, the app would take 2 minutes to read all the files on start up.

Anyway, since we start with .eng/.rsp files we are limited with the data we have - and sometimes it's contradictory or just wrong. Thrustcurve has its own database which has things like "corrected mass" and "case info" and "propellent type" along with references to the .rsp/.eng files.  It isn't too hard to access this through the xml/http interface.

Yes, I think it is worth while to do this since we can have more information (case info allows us to start working on the CTI & AT spacer masses).

teyrana, 2015-09-30:
For clarification, what does the extra data give us?  Accuracy?  Reliability? New options? 

Data is just bits unless we can use it....

kruland2607, 2015-09-30:
thrustcurve maintains a lot of extra data which is not available in the .rsp/.eng files - this includes 

1) corrected dimensions & mass 
2) case for reloads (needed if OR wants to model "adapter" kits) 
3) propellant type 

It also allows for complete & correct correlation of different burn profiles (samples) to the motor name. Right now, this is a little bit of a guess in OR and it has been reported by users that some motors are duplicated in the list. 

Kevin

wcravens, 2015-09-30:
I added: "E.g. We report motor type as 'Unknown' when we have an eng file but not an rse file from ThrustCurves. But ThrustCurves does have this data available, we're just not using it."

To the initial comment.