openrocket/openrocket #247: Convert ThrustCurve engine info to SQLite database

wcravens, 2015-09-30:


teyrana, 2015-09-30:
Suggested starting point: the included SQL & Database classes included in the standard API:
[1] API
http://docs.oracle.com/javase/8/docs/api/java/sql/package-summary.html
[2] Tutorial
https://docs.oracle.com/javase/tutorial/jdbc/

kruland2607, 2015-09-30:
Java DB (oracle's implementation of derby) seems like a good candidate since it's included in the jdk and provides for r/o databases distributed in jars and loaded from classpath.

http://docs.oracle.com/javadb/

bkuker, 2015-09-30:
I'd love to see motor loading that could take into account changes to the
motors at run time... If it were possible to reference a directory of eng
files, and load new ones as they appear and reload them as they change,
re-running any simulations that use them or at least marking the
simulations out of date, that would be amazing. I am not sure how many
other users would like that... I usually stop and start OR about 100 times
in a sitting because I am dialing in a motor design on a completed rocket.

-Bill

kruland2607, 2015-09-30:
Bill, thanks for bringing this up.  It impacts this idea in a few ways - how do the current "user defined motors" interact with an embedded sql db?  Could derby's r/o database still support inserts (which only live through a single execution)?  Directory watching is much easier now and could be added to the current infrastructure without much issue, but we'd need to have a new event for "motor changed".

bkuker, 2015-09-30:
I'd just imagine that the embedded sql db and the directory db are
different implementations of a "motor db"... It was really the lack of any
sort of "motor changed" event in OR that prevented me from just doing it.

I actually integrated the OR core into my motor application, it re-runs
every simulation in the ORK file after doing an ugly hack to replace *all*
motors with the latest version of the motor I am editing... Even the OR
core makes it very hard to change a motor at runtime. This is my
implementation of an OR motor database that I use; no matter what motor you
ask for it returns the one you are editing:

https://github.com/bkuker/motorsim/blob/master/gpl/com/billkuker/rocketry/motorsim/visual/openRocket/OneMotorDatabase.java

-Bill

closed 2025-12-24 by SiboVG