openrocket/openrocket #250: UI doesn't scaler to Higher resolutions - Convert Icons to Vector Graphics

teyrana, 2015-11-16:
OpenRocket UI doesn't scale well to High DPI Screens. 

Some text is scaled by the OS (Win 10) text and visibility options - but this has the effect of mangling the parts tree - the fonts are bigger, but not spaced vertically properly. Some fonts (like in the rocket view window) aren't scaled at all.  This is most acute in the new component icon graphics, but likely will manifest all over the program.

    The default resolution is 2736x1824, and I'm thinking that OpenRocket is just not scaling well to the hi DPI. 

Reported by:
    Charles McGonegal (via The Rocketry Forum)
Edited by:
   Daniel Williams

teyrana, 2015-11-24:
NB:  this looks like it would require an external library to render .svg files.   Batik seems like the de facto standard at this time: https://xmlgraphics.apache.org/batik/

matt-kendall, 2016-11-05:
I don't know much about the rendering / resolution side of things (and I don't have a 4K screen to test on!) but I would be happy to take on the vectorization of icons / Batik side of things.

kruland2607, 2016-11-06:
The icons were created in gimp with original gimp multi-layer sources 
files in core/resources-src/pix.  I would love for these to be svg so 
they can scale better.

matt-kendall, 2017-01-15:
Original discussion on TRF: http://www.rocketryforum.com/showthread.php?129996-Open-Rocket-on-hi-dpi-screen-any-way-to-zoom

plaa, 2017-01-15:
Note that the disabled versions are implemented by removing every second
pixel (in checkerboard style) iirc in the Java code. I originally tried
making them partly transparent / grayed out, but it just didn't look
disabled to me.

- Sampo J.

showthread.php?129996-Open-Rocket-on-hi-dpi-screen-any-way-to-zoom

snrkl, 2018-03-25:
I am also happy to work on vector icons if assistance is needed.. I am always looking for small projects like this, and given I live in OR, I would be happy to contribute...

teyrana, 2018-03-25:
Well, those definitely need work! I'd love to see those upgraded :)

SpamWall, 2018-03-25:
So where is the latest development going on?  GitHub openrocket/openrocket?  Didn’t someone create a new fork for active development because no one was merging changes on openrocket/openrocket?

wcravens, 2018-03-25:
> AEDWd72eOUeVKznvFyLeVr05Wy_cDr9iks5th_aZgaJpZM4GjIZ-> . <

openrocket/openrocket 'unstable'
<https://github.com/openrocket/openrocket/tree/unstable> is effectively the
conical development head.  Once we release we can decide what to do about
things in master that are not in unstable.

I sure hope no-one forked.

Wes

Bjohnson131, 2024-06-25:
+1 to this issue. OpenRocket is almost unusable on 4k.

SiboVG, 2024-06-25:
> +1 to this issue. OpenRocket is almost unusable on 4k.

You know you can change the UI text size since OpenRocket 23.09 (Preferences > General)? It doesn't scale the icons, only text, but it will fix your issue of the program not being legible.

SiboVG, 2025-02-09:
#2628 incorporates UI scaling.

closed 2025-02-09 by SiboVG