openrocket/openrocket #282: UML diagram

Vicilu, 2016-10-04:
Hello, I'm currently making an uml diagram for my understanding of the code, and I gotta ask, besides the rocket component tree in the doc, is there any other UML diagram?
What I'm saying will it be worth cleaning and arranging all those lines?
Anyway, UML diagram for the code coming soon! 
[image]

SpamWall, 2016-10-04:
You work is appriciated.

Sent from TypeApp

teyrana, 2016-10-05:
> ![umlgrande](https://cloud.githubusercontent.com/assets/
> 14062008/19091925/ee515e80-8a5a-11e6-9241-4d3f09a9674d.png)

>   is there any other UML diagram?

Nope. Not really.  It's such a large code base that putting every class in
the same diagram would be insane, and honestly too complex to really
provide an advantage over the code itself.

> What I'm saying will it be worth cleaning and arranging all those lines?

Not Exactly.  BUT, there would DEFINITELY be value in hand-curating subsets
of the mast diagram to highlight certain use cases.  For instance:
- how to calculate the C.G. of a rocket or stage?
- how to simulate a rocket?

Just my humble opinion, of course.  Making the code more accessible would
surely encourage others to get more involved, right?

Cheers,
-Dan

Vicilu, 2016-10-05:
yes, that's my view of it,
I'm part of a rocket research group and they constantly make new programs doing the same thing, but somethings are things that openrocket doesn't do, others are inspired by the openrocket documentation of calculations.
 One said, "we probable could write a better program than openrocket",
 I said "why not add those things into openrocket?" 
But when I got here I was faced with lots of classes, but little documentation, would take a long time to read it all and longer to understand, so I'll spend this time clearing the diagram (it's working), that's why I'm doing it, many of our group doesn't know java and a diagram would certainly help on that. The changes will come in time

teyrana, 2016-10-05:
> I'm part of a rocket research group and they ....

Wait, wait, wait, wait. WAIT.  You can't just throw that out in passing!
What do you guys do?  What are your research topics? Is it more theoretical
or experimental?
Whatever it is, it sounds cool!

Cheers,
-Dan

p.s. If you have specific questions about classes, feel free to ask on
here.  I'll pipe in if I know.  (Although traffic / replies can otherwise
be rather light on here...)

Vicilu, 2016-10-08:
hahaha
I'm part of group TOPUS of aerospace research from the University of São Paulo, Brazil, we are mostly undergrad students, with some in the masters degree.
We are making a rocket for competition in the IREC 2017, but we have a lot limitation in term of rocket market (it's just one vendor, with very limited resource), so we rely a lot in simulations before creating anything, We have some papers published on some topics, but are in portuguese

teyrana, 2016-10-10:
Ahhhh, sounds like you guys are most interested in building the rocket,
huh?

Well sounds cool, regardless :)

(And you're right -- I don't know portuguese, sooo.... point taken.)

Vicilu, 2016-10-16:
so... I've been documenting the code so I can understand, and I've passed through the communication package, there's an UpdateInfoFetcher object that is used in UpdateInfoRetriever, but that's not in the code, am I missing something in my files?

kruland2607, 2016-10-16:
UpdateInfoFetcher is an inner class of UpdateInfoRetriever.

Vicilu, 2016-10-16:
oh yes, it was a problem with my brackets, thanks

dkingsley, 2017-06-20:
I think some well done UML diagrams for various aspects of OR are necessary but should not hold up the promotion of unstable.

wolsen, 2020-05-24:
I believe that this is generally handled with PR #639 where Sourcespy is used to generate the UML diagram automatically.

BigsonLvrocha, 2020-05-25:
this is old, so I'm closing

closed 2020-05-25 by BigsonLvrocha