openrocket/openrocket #296: bug when undoing the first component

Vicilu, 2016-10-19:
-open program
-add nose
-undo
-get bug

I tried with the unstable and master, master doen't give error

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NullPointerException
	at net.sf.openrocket.aerodynamics.BarrowmanCalculator.calculateNonAxialForces(BarrowmanCalculator.java:192)
	at net.sf.openrocket.aerodynamics.BarrowmanCalculator.getCP(BarrowmanCalculator.java:64)
	at net.sf.openrocket.aerodynamics.AbstractAerodynamicCalculator.getWorstCP(AbstractAerodynamicCalculator.java:64)
	at net.sf.openrocket.gui.scalefigure.RocketPanel.updateExtras(RocketPanel.java:597)
	at net.sf.openrocket.gui.scalefigure.RocketPanel.access$7(RocketPanel.java:563)
	at net.sf.openrocket.gui.scalefigure.RocketPanel$6.stateChanged(RocketPanel.java:278)
	at net.sf.openrocket.util.ComponentChangeAdapter.componentChanged(ComponentChangeAdapter.java:20)
	at net.sf.openrocket.rocketcomponent.Rocket.notifyAllListeners(Rocket.java:491)
	at net.sf.openrocket.rocketcomponent.Rocket.fireComponentChangeEvent(Rocket.java:461)
	at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:1738)
	at net.sf.openrocket.rocketcomponent.Rocket.loadFrom(Rocket.java:372)
	at net.sf.openrocket.document.OpenRocketDocument.undo(OpenRocketDocument.java:532)
	at net.sf.openrocket.gui.main.UndoRedoAction.actionPerformed(UndoRedoAction.java:64)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.AbstractButton.doClick(Unknown Source)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]
[... 351 lines cut ...]
 360    101,885 STDERR (SwingExceptionHandler.java:49) 	at java.awt.Window.dispatchEventImpl(Unknown Source)
 361    101,885 STDERR (SwingExceptionHandler.java:49) 	at java.awt.Component.dispatchEvent(Unknown Source)
 362    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
 363    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue.access$500(Unknown Source)
 364    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$3.run(Unknown Source)
 365    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$3.run(Unknown Source)
 366    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.security.AccessController.doPrivileged(Native Method)
 367    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
 368    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
 369    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$4.run(Unknown Source)
 370    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$4.run(Unknown Source)
 371    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.security.AccessController.doPrivileged(Native Method)
 372    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
 373    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue.dispatchEvent(Unknown Source)
 374    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
 375    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
 376    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
 377    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
 378    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
 379    101,886 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.run(Unknown Source)
 380    101,886 INFO   (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
 381    101,886 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 382    104,923 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 383    104,927 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

teyrana, 2016-10-19:
Well THAT's new. Thanks for the heads up!

It looks related to the known integration issue, so we'll keep an eye out--
hopefully they're related.

teyrana, 2016-10-22:
This issues looks to be have the same cause as Issue #294.  I'll be adding my comments there.

Vicilu, 2016-10-26:
it was from a new rocket, but since the merge and the bug fix, it has been working, so I'll close the issue

closed 2016-10-26 by Vicilu