openrocket/openrocket #300: Refactor Rocket Motor Configuration classes into new package structure.

SpamWall, 2016-10-23:
The current GUI package can use some refactoring to provide better modularization.

teyrana, 2016-10-25:
Most of us agree that is true-- but here's the catch: refactoring tends to
introduce bugs.

And like it or not,  ugly but working code is for preferable to pretty
broken code.  (no matter how much it bugs you to look at it.)

BUT, Having said that . . .

Let's circle back after the next release, and we can consider *specific*
design changes.  Having attempted to "fix" some of those changes myself,
it's more complicated than you'd expected.

-DMW

-- 
<a href="https://github.com/teyrana
<https://www.linkedin.com/in/danielmcclinticwilliams>"><img src="
https://static.licdn.com/scds/common/u/img/webpromo/btn_viewmy_160x25.png"
width="160" height="25" border="0" alt="View Daniel Williams's profile on
LinkedIn"></a>

<a href="https://www.linkedin.com/in/danielmcclinticwilliams"><img src="
http://www.flaticon.com/free-icon/github-logo_25231
<https://static.licdn.com/scds/common/u/img/webpromo/btn_viewmy_160x25.png>"
width="64" height="64" border="0" alt="Github.com"> Github Profile </a>

Vicilu, 2016-10-26:
for this reason I've been searching for automated functional tests for swing, I think we should be doing this, btw, just noticed you have the test, but it's nop, that bug I found on the previous issue, where I created a nose and undo giving bug could have been found earlier with those test, this way we won't be afraid of 
Using the next release as a base for writing those would be a great idea, then we won't be afraid of refactoring, and that would be great :D because I've been doing refactoring and sometimes I miss being able to just push a button to see there was no bug introduced

SpamWall, 2017-06-20:
If "refactoring tends to introduce bugs" then the test coverage is not sufficient.  The purpose of jUnit tests is to enable refactoring.

closed 2022-03-22 by neilweinstock