openrocket/openrocket #304: Preferred units not used by offset or angle for Pods and Boosters.

matt-kendall, 2016-11-05:
My preferred units for distance and angle are set in preferences as mm and degrees:
[image]

However, in the configuration for pods or boosters, the 'offset value' and 'angle' fields are given in cm and radians respectively:
[image]
[image]

matt-kendall, 2016-11-05:
PR 306 opened to fix this

teyrana, 2016-12-11:
PR #306 has been accepted and merged into openrocket/unstable. Recommend we close as this issue as resolved.

matt-kendall, 2016-12-11:
Closed, I've checked out openrocket/unstable and verified the fix.

closed 2016-12-11 by matt-kendall