openrocket/openrocket #305: Tube fins icon has visible background

matt-kendall, 2016-11-05:
Most icons in the 'add new component' panel have no visible background or border against the button, e.g. this body tube button:
[image]

However a visible outline of the background is visible on the icon for tube fins in all 3 button states:
[image]
[image]
[image]

matt-kendall, 2016-11-05:
I've created PR 305 to fix this for now (I set the transparency in the image).

Is the process for icon creation documented anywhere? I see we have resources-src and resources folders and some icons are missing from src (boosters, pods, stages). I also notice that some icons differ between 'small' and 'large' versions (e.g. boosters). Do we have things like required resolutions documented anywhere? This may all change if issue 250 is completed I suppose: https://github.com/openrocket/openrocket/issues/250 

Also noticed that the icon for tube fins looks very similar to the icon for a launch lug:
[image][image]
Some possible solutions might be to make the tube fin look a little bigger in the icon or redesign it, perhaps with a front view that looks something like this (a front view like the pods icon):

[image]

kruland2607, 2016-11-06:
The reason the tube fin icon looks like the launch lug icon is because I 
created the tube fin one from the launch lug.  I don't know anything 
about gimp/graphics and it was the best I could do. We don't have any 
requirements documented for resolutions and probably they need to be 
completely reworked since they are having difficulty keeping up with 
technology.

teyrana, 2016-12-11:
This looks like it was fixed by commit 3642398b1a5ea9e59ce91f4b96b31c3f0f816976
Any objections to closing this this Issue as resolved?

matt-kendall, 2016-12-11:
Yes, I'll close this - I've checked out the latest unstable and verified the fix

closed 2016-12-11 by matt-kendall