openrocket/openrocket #308: JDK regression: JavaScript not supported as scripting language

plaa, 2016-12-07:
The Oracle JDK and OpenJDK comes built-in with support for JavaScript scripting.  When I created the scripting support in the simulation extensions, JavaScript was intended to be supported on all systems by default.  Now when I start OpenRocket on Linux Mint 18 running OpenJDK 1.8.0_111 no scripting languages are supported.  I believe this is caused by changing from OpenJDK 7 to 8 (at which point the Rhino engine was changed to the Nashorn engine as well).

For debugging I added the following code to both OpenRocket.java and SwingStartup.java:

		ScriptEngineManager mgr = new ScriptEngineManager();
		List<ScriptEngineFactory> factories = mgr.getEngineFactories();
		
		for (ScriptEngineFactory factory : factories) {
			System.out.println("ScriptEngineFactory Info");
			String engName = factory.getEngineName();
			String engVersion = factory.getEngineVersion();
			String langName = factory.getLanguageName();
			String langVersion = factory.getLanguageVersion();
			System.out.printf("\tScript Engine: %s (%s)%n", engName, engVersion);
			List<String> engNames = factory.getNames();
			for (String name : engNames) {
				System.out.printf("\tEngine Alias: %s%n", name);
			}
			System.out.printf("\tLanguage: %s (%s)%n", langName, langVersion);
		}

It's evident that initially when starting OpenRocket JavaScript is supported (in OpenRocket.java) but after the startup classpath changes (in SwingStartup.java) the support is no longer present.  I'm unsure why the classpath updates no longer keep the default nashorn.jar on the classpath or how to fix the issue.

As a workaround, copying `/usr/lib/jvm/java-8-openjdk-amd64/jre/lib/ext/nashorn.jar` to `~/.openrocket/Plugins/` fixes the issue, as this forces `nashorn.jar` to be on the updated classpath.

teyrana, 2016-12-07:
Ummm, we use Javascript? For what?

plaa, 2016-12-07:
In simulation extensions.  Open the example design "Simulation extensions and scripting", then double-click the "Active roll control" simulation, select Simulation options and click the wrench on the "JavaScript script" extension.  Really powerful stuff when you need something custom or want to simulate flight computers.

dkingsley, 2017-06-20:
Seems to work on Win10 with Oracle Java 8 update 131. I do not have a Linux machine with a Java Dev environment.

plaa, 2017-06-20:
Problem occurs on Mac OS Sierra with Oracle Java 1.8.0_121.

teyrana, 2017-06-21:
Reproduced on `unstable`, @a59aadbf
Mac OS 10.12.5, with Oracle Java 1.8.0_31  :    It runs, but the simulation is broken.
Linux: OpenJDK 1.8.0_131:   Fails to simulate with an exception.

So, it definitely exists, but is it our bug, or an OpenJDK "bug"?

plaa, 2017-06-21:
It's an artifact from the jar-in-jar classpath mungling using Java 8.
Before the classpath update the scripting engine is available, after the
update it's not.  Not sure whether it could be classified as a JRE bug
(i.e. is the classpath mungling something that should work on all Java
implementations), but we need to deal with it anyway.

Don't know how to work around it though.

thzero, 2022-03-04:
Yup.  This one is still around.  IDE it works fine.  Even in the unit test is working because it does not use the jar-in-jar classpath mungling.  However, the  jar-in-jar classpath mungling does not allow the scripting engine to be available; it seems to be using the application .jar file on the classpath instead of the individual .jar files that the IDE is using.

PR #1507 (2022-08-02): Resolving issue #1270, unexpected exception with simulation scripting
This PR fixes #1270, fixes #308, fixes #1108 and fixes #826.

As the Nashorn scripting engine is deprecated, as part of the resolution this replaces Nashorn with the Graal.Js scripting engine.  This includes the addition of the following dependencies

graal-sdk-22.1.0.1.jar
icu4j-71.1.jar
js-22.1.0.1.jar
js-scriptengine-22.1.0.1.jar
truffle-api-22.1.0.1.jar
closed 2022-08-02 by SiboVG