openrocket/openrocket #309: Regression: Motor mass does not decrease during flight

plaa, 2016-12-08:
In OpenRocket 15.03 the mass of a rocket does not change during a flight simulation.  In OpenRocket 14.11 it decreased as the motor burned.  This causes too low estimated apogee altitudes.  I've verified that this affects the simulation results and is not just a problem of storing the mass value.

teyrana, 2016-12-08:
Hmm, you're referring specifically to the release versions (15.03, 14.11)?

I submitted a bugfix for this in the current development version,  so let's
verify that it's the same bug.  Would you please send me the file you're
seeing this in?

Thanks,
-Dan

plaa, 2016-12-08:
Sorry, my bad.  I was running a development build, which I believe was forked from an old revision of the unstable branch.

I'll leave this open until it's verified this no longer occurs in the unstable branch.  Feel free to close this if you know it's working.  Simulating any rocket and plotting the rocket mass showed a constant mass.

teyrana, 2016-12-11:
Verified as of 12/10/2016, and commit 597d6c8 (unstable branch)

closed 2016-12-11 by plaa