openrocket/openrocket #323: "0 degrees" on pods should always extend radially away from the design center

neilweinstock, 2017-02-03:
I was trying out the latest unstable release to try out the pods implementation and was alarmed at how 0 degrees on all the pods always pointed straight up, for example:
[image]

I don't know if this is a bug or by design but it really should be changed to make the feature useful.  Zero degrees should always point away from the center of the design, so that if I add a single fin to a set of three pods it looks like this:
[image]

Creating this in the current version is... well let's just say "unpleasant".

I sincerely hope this can be fixed before the next release.  Thanks for considering!

teyrana, 2017-02-04:
Thanks for the heads up.  I wrote that code, and what you describe was definitely NOT intended behavior. I'll look into it when I get a chance.

neilweinstock, 2017-02-04:
Excellent, thanks.  I was sure hoping that wasn't what was intended. [😊]

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

ChrisMickelson, 2017-02-08:
FYI, I've noticed the same behavior on parallel staging/boosters when a small fin is added to the booster tube.

ChrisMickelson, 2017-06-30:
I'll add that perhaps there could be a check-box to align pods radially or unidirectional. For example, pods with fins like Neil mentioned above, or airplane-style pods where you would want vertical stabilizers unidirectional.

neilweinstock, 2017-06-30:
Glad you mentioned that, Chris, since I've been thinking about exactly that issue.  It mainly occurs on two-pod systems, where you'd very commonly want something along these lines:
[image]

0 degrees can still extend radially out from center BT (as shown in the picture), or it can stay vertical; either way works.  The important thing is that the rotational direction of the fins on each pod should be opposite.  This option should apply *only* to two-pod sets; otherwise it doesn't really make any sense.

In fact, in the absence of a selectable option, I'd actually make two-pod sets work like this by default, since it seems *much* more likely to be useful than the other behavior..

Without this capability, entering a design like the one in the picture would require two separate single pods each with its own copy of the fins, etc.

teyrana, 2017-06-30:
@dkingsley Please assign this issue to me.

teyrana, 2017-10-14:
Fix pending at:
https://github.com/teyrana/openrocket/tree/rc1

commit c36ce2ea (2018-01-06): [resolves #323] Major patch to improve rendering of instanced+rotated components

- introduced interfaces for different types of positionable components:
  -- AnglePositionable
  -- AxialPositionable
  -- RadiusPositionable
- RingInstanceable now includes these other interfaces, making explicit the expectations of
   any implementing subclass.
- RingInstanceable now indicates its angles as an array instead of returning angles one-at-a-time
  -- commit includes updates to match this new API
- Added getAssembly() method to RocketComponent, to simplify instancing code
closed 2018-01-22 by wcravens