openrocket/openrocket #325: Repeatable crasher on Mac when using 3D views

neilweinstock, 2017-02-10:
This problem has been around a while, doesn't seem to be fixed in the current unstable build so I'm reporting in hopes it can be fixed before the next release.  It only manifests on the Mac; I'm not sure it happens on *everyone's* Mac but a number of people on TRF confirmed that they have the same problem, so it's not only me.

Whenever you select any of the 3D views (Figure, Unfinished, or Finished) it works fine, but then switching back to one of the 2D views *or* quitting the program while the 3D view is still up will cause OR to crash.  Sample log file is attached.

In my testing of the latest version I have, it didn't crash *every* time I switch to a 2D view, but if I switch back and forth a couple times it will crash pretty quickly.

Thanks,
Neil

[hs_err_pid40821.txt](https://github.com/openrocket/openrocket/files/765348/hs_err_pid40821.txt)

wcravens, 2017-02-10:
Very cool reports Neil.  I just wanted to say thank you for your efforts to help by reporting these things.  We're a pretty quiet list but I promise that folks are paying attention and that your work is adding value.

Thanks, 

Wes

neilweinstock, 2017-02-10:
Glad you like, there's more coming. :)

- Neil

—

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

neilweinstock, 2017-10-28:
So it appears that checking "Use off-screen rendering" fixes this, at least for the few who've tried it so far.  Is there any disadvantage to using off-screen rendering, any reason it's not checked by default?

teyrana, 2017-10-28:
@neilweinstock  @wcravens 

Can you guys post which video cards you've found this issue on? Just to see if that's a related issue, or not...

neilweinstock, 2017-10-29:
As requested by Daniel: my machine is an iMac with NVIDIA GeForce GT 750M 1024 MB.  However, anecdotally, pretty much every Mac user I've heard from has this problem, so I don't *think* it's graphic card-specific, but I could be wrong.

teyrana, 2017-10-29:
Well apparently, I'm the 1/10000 mac user who doesn't get the problem.....
Because the 3d views arent working for me *at all* :/

snrkl, 2017-10-29:
for what it is worth, it would seem that off-screen rendering fixed the issue for me also on two machines:

1) MacBook Pro (Retina, 15-inch, Late 2013) 2.6 GHz Intel Core i7, 16 GB 1600 MHz DDR3, NVIDIA GeForce GT 750M 2048 MB running OSX El Capitan 10.11.6 (15G1611)

2) MacBook Air (11-inch Mid 2013), 1.3Ghz Intel Core i5, 4GB 1600MHz DDR3, Intel HD Graphics 5000 1536MB, running OSX Sierra 10.12.6 16G29

wolsen, 2020-04-26:
Is this fixed in the latest unstable builds? The jogl libraries have been updated which handles some known mac os x bugs. If this can't be reproduced on unstable, I'd like to close this one out.

neilweinstock, 2020-04-26:
I haven’t gotten a jar file from the mainline unstable in a long time.  If I can get one I’ll be happy to double-check this one, and some others as well.

  *   Neil

wolsen, 2020-04-27:
Ask and ye shall receive:

https://www.dropbox.com/s/mse2igh2ngba4yz/OpenRocket-unstable-1c7171ef.jar?dl=0

Its the latest from unstable, based on commit 1c7171ef

neilweinstock, 2020-04-27:
Thanks, most helpful.

I cannot reproduce this error with the latest build you provided.  Tried with off-screen rendering disabled.

  *   Neil

https://www.dropbox.com/s/mse2igh2ngba4yz/OpenRocket-unstable-1c7171ef.jar?dl=0<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.dropbox.com%2Fs%2Fmse2igh2ngba4yz%2FOpenRocket-unstable-1c7171ef.jar%3Fdl%3D0&data=02%7C01%7Cneil.weinstock%40raritan.com%7C882faaa38e674121ac6808d7ea5fe279%7C199686b5bef4496087867a6b1888fee3%7C1%7C0%7C637235569852954083&sdata=rIIWvYDdGD1X8zS%2FAx56IPEo5XFF5mVyQ832A9YsxOM%3D&reserved=0>

neilweinstock, 2020-05-01:
I guess I shall close this one, seems to be fixed in latest unstable build.

closed 2020-05-01 by neilweinstock