openrocket/openrocket #328: Pods not saved properly when #copies = 1

neilweinstock, 2017-02-24:
K'Tesh raised this issue on the mailing list but I don't believe he filed it here, so I'll do it for him.

In the Pods/Parallel tab, up/down control on "Number of Copies" won't go below two (why?) but it works fine to manually set the value to one.  It then seems to be working perfectly fine until you save it.  Then, when you re-read the file and look at the pod set again, you will see:

1) Number of copies has been set back to "2".
2) Radial Distance and Angle have both been zeroed out.

It will be a severe limitation if number of copies is not allowed to be set to "1", so I'm hoping this can be fixed.

teyrana, 2017-02-25:
The limit-to-2 decission was based on the assumption that those pods are
external to the main stack, and is directly tied to the underlying
aerodynamic assumptions.

The design use case was an external strapon boosters, like the delta 2 or
Soyuz.  Aerodymically, it greatly simplifies the calculations, because the
torques are balanced.  If you have a single pod that's offset, now you have
an unbalanced torque that will quite literally make your pod unstable.

If you're putting a pod inline with above stages, the code will
overcalculate your drag, (as if that nose cone was exposed to freestream
airflow.)

So, unfortunately, it's not as simple as changing the min bound on the
count.
Can you elaborate on what you're trying to do with the single pod?

neilweinstock, 2017-02-27:
Hi,
First of all let me say I greatly appreciate the opportunity to discuss these design issues.

You wrote:
The design use case was an external strapon boosters, like the delta 2 or
Soyuz. Aerodymically, it greatly simplifies the calculations, because the
torques are balanced. If you have a single pod that's offset, now you have
an unbalanced torque that will quite literally make your pod unstable.

That’s a very narrow use case.  Pods, like almost everything in OR, *will* get used for lots more than their original intended purpose (example: internal tubes as tail rings, given the lack of explicit support for tail rings).  In particular, pods give us our long-awaited means to implement fins-on-fins.  I’m a little unclear though: when you talk about external strap-on boosters and balanced torques, are you talking about pods with motor mounts in them, or are you just talking about torque generated by asymmetric drag?

If the latter, then it would seem that asymmetric drag already must be handled (or ignored, as appropriate) by the program since I can already create all kinds of asymmetry in the design without pods.

If it’s actually about unbalanced motors, though, then it’s even more of a niche use case; I can think of far more pod-based designs without motors in the pods than with.  In that case, though, there would seem to be a straightforward solution: don’t allow motor mounts in single-instance pods, or conversely don’t allow pod instances to be set to “1” in pods that have motor mounts.  Admittedly this requires a few extra checks and coding to implement (it’s not free), it’s not awful, and it would seem to eliminate the problem, *if* that is indeed the problem.

If you're putting a pod inline with above stages, the code will
overcalculate your drag, (as if that nose cone was exposed to freestream
airflow.)

There are already lots of ways to confuse the simulation without resorting to pods; prohibiting all those cases would be almost impossible and would in fact make the program much less useful.  It is simply incumbent on the user to understand what the heck they’re doing.

I already know that any time I put a faux-tail ring into a design, I’m going to get inaccurate CP and drag results.  Likewise, phantom body tubes to attach tube fins to fins (rather than to the main BT).  There are others.  In those cases, I’ll often create a secondary sim file that eliminates all the confusing and/or unsimmable elements, replacing them with others that I hope are well-handled in the sim, and roughly equivalent in terms of their aerodynamics.  Ya do what ya can.

[As an aside, this is why I have wished for, although I haven’t formally requested, the ability to tag components as “no-sim”; these would be either purely for visualization, or for things like phantom body tubes that are there just to support other components.  In the current version, phantom body tubes tend to generate “diameter discontinuity” warnings despite the fact that they’re not actually there].

So, unfortunately, it's not as simple as changing the min bound on the count.
Can you elaborate on what you're trying to do with the single pod?

In the particular design I’ve been working on recently, I was using single-instance pods purely to work around my previously reported issue that fin angles did not properly rotate with the pods.  However, there a thousand examples of rocket designs, usually airplane-styled, which are inherently non-radially symmetric:

1)      I could make a model of the Cosmic Interceptor (well, most of it, anyway), using pods to create the fins-on-fins on the underside<http://img.auctiva.com/imgdata/1/5/1/7/6/2/0/webimg/781143487_tp.jpg>.

2)      I could totally do a model of the Estes Lynx<http://www.estesrockets.com/media/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/7/2/7233_lynx_feature_2.jpg>.  I’d need a single instance pod on top to mount the top front wings, and one on the bottom for the bottom rear wings.  A symmetric set of 2 pods could implement the winglets.

3)      Someone just did a build thread of the old Estes Nova Scout<http://i1122.photobucket.com/albums/l529/rmortis/001_zpsbac27f27.jpg>.  That would require a single pod up top to implement the fins on the tube.

4)      The Estes Andromeda<http://www.spacemodeling.org/jimz/estes/est1273@.gif> likewise requires a single pod up top to implement the cross-piece.

5)      How about the original Interceptor<http://archive.rocketreviews.com/images_est/estes_interceptor_est.jpg>?  The wing pods, would need to be entered as two individuals.

6)      Heck, I could even go completely nuts and do the Orbital Transport<https://www.rocketreviews.com/images4/scratch_4x_orbital_transport01.jpg>.  I wouldn’t have too much expectation that it would actually simulate anywhere close to realistically, but I could absolutely enter it and get a CG if nothing else.

7)      Or the Orbital Interceptor<http://www.rocketryforum.com/attachment.php?attachmentid=5901&d=1242485351>, the Estes Attack Craft Orion<https://www.flickr.com/photos/wallyum/sets/72157600220653124/>, the North Coast Hyperswift<https://www.apogeerockets.com/bmz_cache/5/5cc25c515ab7fc14f4d3b70797b3183f.image.320x320.jpg>, or…..

You get the idea.  Although the current implementation of pod sets, as children of the sustainer, makes it a bit challenging to enter some of these designs, it’s still *possible*.  That is, if and only if we can do a single-instance pod.

Thanks for listening!

[... cut ...] tion ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2017-03-01:
> om/media/catalog/product/cache/1/image/9df78eab33525d08d6e5f
> b8d27136e95/7/2/7233_lynx_feature_2.jpg>. I’d need a single instance pod
> gif> likewise requires a single pod up top to implement the cross-piece.
> ketreviews.com/images_est/estes_interceptor_est.jpg>? The wing pods,
> attachmentid=5901&d=1242485351>, the Estes Attack Craft Orion<
> 15ab7fc14f4d3b70797b3183f.image.320x320.jpg>, or…..

So, basically it boils down to looks vs accuracy.   (This came up during
the debate on tube fins, as well-- What was the agreement we came to last
time?

Probably the easiest solution would just be to flag the entire rocket as
un-simmable.  ( Either manually, or upon detection of certain features?)

Personally I'm not comfortable having a 'simulation' tab that will happily
produce simulations even when we know the results will be crap.  Not as a
matter of principle, but as a matter of User Experience.

KTesh, 2017-03-01:
Here's the problem...  Besides creating simulations for flight characteristics, OR is also a major tool for cloning OOP designs.  

Not all pods can be/should be used for motors.  Not all rockets that had them, had motors in them.  There are cases where rockets had offset motors...  Boost gliders come to mind.  I don't need to know what they'll sim to, I just want to preserve rocketry history.  I'll trust that the designers from Estes, Centuri, or wherever else designed their kits did so knowing that the rocket was stable, and would work with the motors they suggest.  That said, how many designs have we seen on rocketreviews's site that can't stand up to scrutiny.  I've looked at several of the designs that have come from there (in .rkt format) that are not accurate at all.  Had I not done some additional research, I'd have built rockets based on those files, and I'd have been disappointed when I had them finished, because something just wasn't quite "right".

Personally, I'm hoping to one day sim the Scorpius, Hyperion, Andromeda, or Cassiopeia (to name a few) from the 1983 catalog...  The year I got started.  Those rockets all have fins on fins, or even tube fins on fins. 

Sometimes you just need to add a pod.  One... and only one.

wcravens, 2017-03-01:
A far better tool for these purposes would be CAD.  Such as the wonderful
and free, Fusion 360.

Wes

KTesh, 2017-03-01:
While I might not be interested in the actual flight characteristics, that doesn't mean that everybody won't be.  CAD programs can't offer that.  Besides, I don't know how I'd share a .CAD file on TRF, TRC, YORF, or anywhere.  I don't know the first thing about them.

teyrana, 2017-03-02:
Minor clarification:  pods without motors are simmed ok, in general.  It's
the corner cases of one-off pods/ boosters which we're debating.

Its not even whether we should "ever" implement said features.... But what
should we include on THIS release.

KTesh, 2017-03-02:
Some other kits I want to dim up are the classic Centuri F-104 Star Fighter.  To do those fins, I need the option for one pod, or another way of putting fins on fins.

neilweinstock, 2017-06-21:
Was there a decision whether or not to support #pods = 1?  I sincerely hope the answer is "yes", although I absolutely understand the push to try to get this release out, and how much work there is left to do.

For now, if the user puts a motor in a single pod, then it would seem reasonable to just put up a red warning (like for other things that won't simulate properly, like jagged fins, body discontinuities, etc.)  I think that's likely more straightforward than trying to actually prevent the user from entering it in the first place.  

The important use case is single pods without motors.

commit 212685b0 (2018-01-06): [resolves #328][save][load] Fixed invalid save & load fields for 'FinSet', 'LineInstanceable' Components

- PodSet may save/load auto-radial-offset
- Fins now save/load
  - 'instancecount'
  - 'angularoffset'
  - 'radialoffset' = 'auto'

- RadiusPositionable interface now enforces 'setAutoRadialOffset(...)'
- implemented RockeComponentSaver#emitInteger
- all LineInstanceable implementers read/write 'instanceseparation'
- [fix] RocketComponentSaver now saves 'instancecount' even if 1==instancecount
closed 2018-01-22 by wcravens