openrocket/openrocket #329: UI listeners not listening

neilweinstock, 2017-02-24:
Many linkages within the UI seem to be broken.  Here are the ones I've noticed so far:

1) Sliders disconnected from values: In *all* the ones I tried, dragging a slider does not cause the displayed value to be updated.  It does work the other way, though: changing the value *will* cause the slider know to be repositioned.

2) Body tube wall thickness and inner diameter not linked: if I modify one of these, the other does not change as it should.  However, changing the outer diameter *does* properly cause the inner diameter to be recalculated given the set wall thickness.

3) Selecting a part from the database: After selecting one, none of the numeric parameter values get updated to reflect the selected part.  The material selector and the "filled" checkbox do however get updated.

Please note that in all cases, the changes are in fact being applied to the component correctly, it is just that the new values are not being displayed in the UI correctly.

ChrisMickelson, 2017-03-06:
For the last few days I've been trying to track down where the disconnect is. Hopefully someone more familiar with the code can help point me in the right direction. So far I'm wondering if it has something to do with a JSpinner not using an adaptor such as DoubleModel or IntegerModel? It appears that everything except the spinner in the JPanel will "hear" the change in value of the slider.

teyrana, 2017-06-03:
@neilweinstock @ChrisMickelson 

I've started to fix these issues, and my branch here:   https://github.com/teyrana/openrocket/tree/events

Your help would be very much appreciated if you can help verify: 
(a) that these fixes solve your problems
(b) a fix here does not solve a event/update problem (and what that problem is)

Thanks!

ChrisMickelson, 2017-06-05:
That's great! Thanks for the update. I spent some time testing and it seems to have fixed the issue #1 above. That part works great. Issues #2 and #3 seem unaffected by this update though. For example, changing the thickness of a body tube wall does not automatically change the value of inner diameter until you click on the slider, then it snaps into place. Loading a part from the database changes everything except the values in the panel. Material type is the only thing that I see updated upon loading of a part. As soon as you start adjusting though, the values snap into place.

teyrana, 2017-06-11:
Update: have resolved issues (1) and (2) for body tubes and inner tubes. (and uploaded to above branch.)

neilweinstock, 2018-08-31:
FYI, I'm still seeing widespread problems here in Daniel's latest alpha 8.

Examples:
1) Selecting a preset from database does not update any fields (tried in body tubes, nose cones, and transitions)
2) Clicking the "automatic" checkbox does not update the relevant diameter value (tried in body tubes, nose cones, and transitions)
3) In body tubes -> general, "Inner Diameter" and "Wall thickness" do not talk to each other
4) In Trapezoidal fin set -> general, "Sweep Angle" and "Sweep Length" do not talk to each other.

That's where I stopped testing for now.  I don't think any sample file is needed, just try any of the above in a fresh design.

teyrana, 2018-09-30:
Todo:
=====

- [ ] BodyTube 
  - [ ] 'Automatic' should update OuterDiameter
  - [ ] 'Filled' should update thickness, InnerDiameter values, and gray-out their editors

neilweinstock, 2018-11-03:
In alpha 9: definitely better, but still issues loading parts from database:

1) For body tubes, selecting an item from the parts database updates everything except the body tube length (shown correctly in side view or 3D view, but left unchanged in Body Tube Configuration

2) For nose cones, selecting a part from the database does not update Nose cone shape or Shape parameter.

3) For transitions, selecting a part from the database does not update Transition shape, Shape parameter, or Transition length

neilweinstock, 2018-12-11:
I haven't been able to break it in Alpha 10.  Tried nose cones, body tubes, and trapezoidal fins.

neilweinstock, 2019-04-18:
The original problem I reported seems to be fixed.

There is a newer problem with listeners regarding the finset editor.  Should I tack that on here, or close this and open a new one?

neilweinstock, 2019-04-18:
OK, new problem (not new to this build, but more recent than this original issue).  It was present in the previous alpha (from February) but I only noticed it now.

In the past, changes to fin shape were reflected *immediately* in the rocket rendering in the main window, in real time.  While dragging a point on the fin outline, you could see it change in the rendering.

Now, it seems that point *moves* are not reflected until at least the edit dialog is closed (or a different component is selected).  On some occasions the changes don't even show immediately then; I have seen it take a very long time for the changes to appear in the rocket rendering (i.e. minutes) but I haven't been able to isolate that particular behavior.

Point deletions and additions seem to cause the rendering to be updated.  So if you add a point and then drag it to the desired location, you'll get an update at the moment of addition, but the subsequent move is not shown.  Likewise, clicking on the fin in the main window causes an update.

Also, and this seems likely to be related, point moves do not register the design as "changed".  So I can move a point in a fin shape and then close the design and it will not ask me to save.

Screenshot shows shape in fin editor not matching the rendering.
[image]

teyrana, 2020-02-17:
This issue is migrated to #555 .

commit 2282d33c (2018-12-21): [fixes #329] refactors some event-handling in component config dialogs
commit fee141d2 (2018-12-21): Merge pull request #494 from teyrana/fix_329_events

[fixes #329] refactors some event-handling in component config dialogs
closed 2020-02-17 by teyrana