openrocket/openrocket #330: Scaling the rocket causes an exception

neilweinstock, 2017-02-24:
If I select Edit -> Scale, I immediately get an exception.  Here's the trace:

---------- Exception stack trace ----------
java.lang.ExceptionInInitializerError
	at net.sf.openrocket.gui.main.BasicFrame$16.actionPerformed(BasicFrame.java:640)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: net.sf.openrocket.util.BugException: BUG: Could not find method getPositionValue([]) from class class net.sf.openrocket.rocketcomponent.RocketComponent
	at net.sf.openrocket.util.Reflection.findMethod(Reflection.java:102)
	at net.sf.openrocket.gui.dialogs.ScaleDialog$GeneralScaler.<init>(ScaleDialog.java:546)
	at net.sf.openrocket.gui.dialogs.ScaleDialog.addScaler(ScaleDialog.java:181)
	at net.sf.openrocket.gui.dialogs.ScaleDialog.addScaler(ScaleDialog.java:172)
	at net.sf.openrocket.gui.dialogs.ScaleDialog.<clinit>(ScaleDialog.java:85)
	... 39 more

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 15.03dev
OpenRocket source: default
OpenRocket location: null
JOGL version: 2.1.5
Current default locale: en_US
System properties:
  awt.toolkit=sun.awt.windows.WToolkit
  file.encoding=Cp1252
  file.encoding.pkg=sun.io
  file.separator=\
  java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
  java.awt.printerjob=sun.awt.windows.WPrinterJob
  java.class.path=C:\Users\neilw\git\openrocket\swing\bin;C:\Users\neilw\git\openrocket\swing\lib\jogl\jogl-all.jar;C:\Users\neilw\git\openrocket\swing\lib\iText-5.0.2.jar;C:\Users\neilw\git\openrocket\swing\lib\jcommon-1.0.18.jar;C:\Users\neilw\git\openrocket\swing\lib\jfreechart-1.0.15.jar;C:\Users\neilw\git\openrocket\swing\lib\OrangeExtensions-1.2.jar;C:\Users\neilw\git\openrocket\swing\lib\jogl\gluegen-rt.jar;C:\Users\neilw\git\openrocket\core\bin;C:\Users\neilw\git\openrocket\core\lib-extra\RXTXcomm.jar;C:\Users\neilw\git\openrocket\core\resources;C:\Users\neilw\git\openrocket\core\lib\opencsv-2.3.jar;C:\Users\neilw\git\openrocket\core\lib\guice-3.0.jar;C:\Users\neilw\git\openrocket\core\lib\guice-multibindings-3.0.jar;C:\Users\neilw\git\openrocket\core\lib\javax.inject.jar;C:\Users\neilw\git\openrocket\core\lib\aopalliance.jar;C:\Users\neilw\git\openrocket\core\lib\slf4j-api-1.7.5.jar;C:\Users\neilw\git\openrocket\core\lib\annotation-detector-3.0.2.jar;C:\Users\neilw\git\openrocket\lib-test\hamcrest-core-1.3.0RC1.jar;C:\Users\neilw\git\openrocket\lib-test\hamcrest-library-1.3.0RC1.jar;C:\Users\neilw\git\openrocket\lib-test\jmock-2.6.0-RC2.jar;C:\Users\neilw\git\openrocket\lib-test\jmock-junit4-2.6.0-RC2.jar;C:\Users\neilw\git\openrocket\lib-test\junit-dep-4.8.2.jar;C:\Users\neilw\git\openrocket\lib-test\test-plugin.jar;C:\Users\neilw\git\openrocket\lib-test\uispec4j-2.3-jdk16.jar;C:\Users\neilw\git\openrocket\swing\lib\logback-classic-1.0.12.jar;C:\Users\neilw\git\openrocket\swing\lib\logback-core-1.0.12.jar;C:\Users\neilw\git\openrocket\swing\resources;C:\Users\neilw\git\openrocket\swing\lib\miglayout-4.0-swing.jar;C:\Users\neilw\git\openrocket\swing\lib\rsyntaxtextarea-2.5.6.jar
  java.class.version=52.0
  java.endorsed.dirs=C:\Program Files\Java\jdk1.8.0_111\jre\lib\endorsed
[... 368 lines cut ...]
1442   1003.227 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
1443   1003.227 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
1444   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$4.run(EventQueue.java:731)
1445   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$4.run(EventQueue.java:729)
1446   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.security.AccessController.doPrivileged(Native Method)
1447   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
1448   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
1449   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
1450   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
1451   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
1452   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
1453   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
1454   1003.228 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
1455   1003.229 STDERR (SwingExceptionHandler.java:49) Caused by: net.sf.openrocket.util.BugException: BUG: Could not find method getPositionValue([]) from class class net.sf.openrocket.rocketcomponent.RocketComponent
1456   1003.229 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.util.Reflection.findMethod(Reflection.java:102)
1457   1003.229 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.gui.dialogs.ScaleDialog$GeneralScaler.<init>(ScaleDialog.java:546)
1458   1003.229 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.gui.dialogs.ScaleDialog.addScaler(ScaleDialog.java:181)
1459   1003.229 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.gui.dialogs.ScaleDialog.addScaler(ScaleDialog.java:172)
1460   1003.229 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.gui.dialogs.ScaleDialog.<clinit>(ScaleDialog.java:85)
1461   1003.229 STDERR (SwingExceptionHandler.java:49) 	... 39 more
1462   1003.229 INFO   (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
1463   1003.229 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
1464   1043.030 USER   (SwingExceptionHandler.java:220) User requested sending bug report
1465   1043.043 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

dkingsley, 2017-06-20:
I can not duplicate on the unstable branch

dkingsley, 2017-06-20:
I was able to duplicate this error on Win 8.1 with Java 8 update 131.

SpamWall, 2017-06-20:
I can duplicated this on unstable.  The problem is that RocketComponent used to have (currently has in master) a method called getPositionValue().  When the menu event tries to display the ScaleDialog, ScaleDialog looks for getPositionValue() using reflection in the RocketComponent class.  Since that method no longer exists in RocketComponent, an exception is thrown.

Why was the getPositionValue() removed?  Is the correct fix to add getPositionValue() back to RocketComponent or rewrite the position code.  I personally don't like the reflection, this should be done through an interface so future developers realize where methods are being called.

wcravens, 2017-06-20:
It was refactored by Dan in commit c1e2ed0c28a7126a54977b6e56b59e6b4f5becde

I'm guessing that at least this one reference was missed.

"[Refactor] Replaced redundant method: 'PositionValue(...)' with 'AxialOffset(...)'"
    
    - These two methods did the same thing, so this patch removes the former from RocketComponent and descendants
    - Wherever used, "PositionValue(...)" was replace with "AxialOffset(...)"
    - Removed some extraneous redirect definitions of PositionValue(...)
    - Fixed some UI references to the old method

SpamWall, 2017-06-20:
Was it switched to "AxialOffset" everywhere?  If not, the scaling code will be a problem as it expects any class that can scale to have such a method.

I think the fix needs to be a "Scalable" interface that forces an interface the ScaleDialog can count on.

Any objections?

⁣Sent from TypeApp ​

teyrana, 2017-06-21:
> 330#issuecomment-309883487

>
> It was refactored by Dan in commit c1e2ed0
>
> <https://github.com/openrocket/openrocket/commit/c1e2ed0c28a7126a54977b6e56b59e6b4f5becde>

"[Refactor] Replaced redundant method: 'PositionValue(...)' with
> 'AxialOffset(...)'"

Yup.  The rationale was that "position" is ambiguous.  Is it in the x,y, or
z direction?  forward? backward?

The 'value' vs 'offset' was meant to communicate that the distance was
relative to another component, as opposed to the origin.

I think the fix needs to be a "Scalable" interface that forces an interface
> the ScaleDialog can count on.

I vote for this, as well.

The current pattern of using reflection to detect these methods is all over
the GUI, and leads to frequent bugs which are not detected until one uses
*that* specific element.

This makes testing much harder, as one doesn't know which methods are
broken until actually using them in the UI. This situation is one of the
things typed languages solve :D

Just fyi, there are other similar situations throughout the GUI, so if this
fix works well, let's keep it in mind for similar situations in future.

Cheers,
-Dan

Daniel Williams
equipoise@gmail.com
858-220-8134 (Mobile)
339-368-6425 (Skype)
LinkedIn <https://www.linkedin.com/in/williamsdm>
GitHub <https://github.com/teyrana>

dkingsley, 2017-06-21:
Issue 331 is essentially the same as this one, I will add a pointer to this issue so duplicate work does not happen.

SpamWall, 2017-06-22:
It seems like this dialog only allows scaling larger.  Is that correct?

dkingsley, 2017-06-22:
in 15.03 the slider ranges from 25% to 400%. I have used it to scale a 4" rocket down to 2".

SpamWall, 2017-06-29:
While I'm in here, would it be OK if I added the ability to scale in each dimension separately?

teyrana, 2018-02-03:
@neilweinstock

We patched this while ago.  Would you please confirm if the fix worked for you? 

 @wcravens FYI

neilweinstock, 2018-02-03:
Just tried it on 18.01 RC4.  It indeed doesn't crash, and seems to scale the rocket correctly.

However, immediately after scaling (at least on the one design I tried), OR is very confused about the dimensions of the rocket.  If I save and then reopen the scaled design, everything is fine.

Example: I have a 27.5" rocket.  I apply 120% scaling to the entire rocket.  Each component of the rocket seems to have been correctly scaled, but in the upper left corner of the display area OR says total rocket length is now 60", and the side view of the rocket is displayed in the right half of the display area.  If I save and reopen the rocket, it reopens correctly, reporting 32.99" total length and fills the display area with the rocket side view.

If this doesn't happen for you, then I'll bundle up a design and some screenshots to show.

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2018-02-04:
Ahhh, yea. I see it. *Of course* it's a event-handling issue. :P

teyrana, 2018-03-03:
Fix implemented; in coord.

teyrana, 2018-03-31:
@wcravens @neilweinstock 
This should be fixed. Recommend we close, or reproduce the error.

commit 28bdfac5 (2018-03-27): [fix][Resolves #330] Rocket size now updates on main screen after scaling rocket
closed 2018-03-31 by wcravens