openrocket/openrocket #331: Scaling a freeform fin shape causes an exception

neilweinstock, 2017-02-24:
Clicking "Scale Fin" in the Freeform fin set "Shape" tab causes an exception.  Here's the trace:

---------- Exception stack trace ----------
java.lang.NoClassDefFoundError: Could not initialize class net.sf.openrocket.gui.dialogs.ScaleDialog
	at net.sf.openrocket.gui.configdialog.FreeformFinSetConfig$1.actionPerformed(FreeformFinSetConfig.java:233)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 15.03dev
OpenRocket source: default
OpenRocket location: null
JOGL version: 2.1.5
Current default locale: en_US
System properties:
  awt.toolkit=sun.awt.windows.WToolkit
  file.encoding=Cp1252
  file.encoding.pkg=sun.io
  file.separator=\
  java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
  java.awt.printerjob=sun.awt.windows.WPrinterJob
  java.class.path=C:\Users\neilw\git\openrocket\swing\bin;C:\Users\neilw\git\openrocket\swing\lib\jogl\jogl-all.jar;C:\Users\neilw\git\openrocket\swing\lib\iText-5.0.2.jar;C:\Users\neilw\git\openrocket\swing\lib\jcommon-1.0.18.jar;C:\Users\neilw\git\openrocket\swing\lib\jfreechart-1.0.15.jar;C:\Users\neilw\git\openrocket\swing\lib\OrangeExtensions-1.2.jar;C:\Users\neilw\git\openrocket\swing\lib\jogl\gluegen-rt.jar;C:\Users\neilw\git\openrocket\core\bin;C:\Users\neilw\git\openrocket\core\lib-extra\RXTXcomm.jar;C:\Users\neilw\git\openrocket\core\resources;C:\Users\neilw\git\openrocket\core\lib\opencsv-2.3.jar;C:\Users\neilw\git\openrocket\core\lib\guice-3.0.jar;C:\Users\neilw\git\openrocket\core\lib\guice-multibindings-3.0.jar;C:\Users\neilw\git\openrocket\core\lib\javax.inject.jar;C:\Users\neilw\git\openrocket\core\lib\aopalliance.jar;C:\Users\neilw\git\openrocket\core\lib\slf4j-api-1.7.5.jar;C:\Users\neilw\git\openrocket\core\lib\annotation-detector-3.0.2.jar;C:\Users\neilw\git\openrocket\lib-test\hamcrest-core-1.3.0RC1.jar;C:\Users\neilw\git\openrocket\lib-test\hamcrest-library-1.3.0RC1.jar;C:\Users\neilw\git\openrocket\lib-test\jmock-2.6.0-RC2.jar;C:\Users\neilw\git\openrocket\lib-test\jmock-junit4-2.6.0-RC2.jar;C:\Users\neilw\git\openrocket\lib-test\junit-dep-4.8.2.jar;C:\Users\neilw\git\openrocket\lib-test\test-plugin.jar;C:\Users\neilw\git\openrocket\lib-test\uispec4j-2.3-jdk16.jar;C:\Users\neilw\git\openrocket\swing\lib\logback-classic-1.0.12.jar;C:\Users\neilw\git\openrocket\swing\lib\logback-core-1.0.12.jar;C:\Users\neilw\git\openrocket\swing\resources;C:\Users\neilw\git\openrocket\swing\lib\miglayout-4.0-swing.jar;C:\Users\neilw\git\openrocket\swing\lib\rsyntaxtextarea-2.5.6.jar
  java.class.version=52.0
  java.endorsed.dirs=C:\Program Files\Java\jdk1.8.0_111\jre\lib\endorsed
  java.ext.dirs=C:\Program Files\Java\jdk1.8.0_111\jre\lib\ext;C:\Windows\Sun\Java\lib\ext
  java.home=C:\Program Files\Java\jdk1.8.0_111\jre
  java.io.tmpdir=C:\Users\neilw\AppData\Local\Temp\
  java.library.path=C:\Program Files\Java\jdk1.8.0_111\bin;C:\Windows\Sun\Java\bin;C:\Windows\system32;C:\Windows;C:/Program Files/Java/jdk1.8.0_111/jre/bin/server;C:/Program Files/Java/jdk1.8.0_111/jre/bin;C:/Program Files/Java/jdk1.8.0_111/jre/lib/amd64;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\Emscripten\node\0.12.2_64bit;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\TortoiseSVN\bin;c:\jruby-9.1.5.0\bin;C:\Windows\system32;;.
  java.runtime.name=Java(TM) SE Runtime Environment
  java.runtime.version=1.8.0_111-b14
  java.specification.name=Java Platform API Specification
  java.specification.vendor=Oracle Corporation
  java.specification.version=1.8
[... 410 lines cut ...]
2422   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.awt.Window.dispatchEventImpl(Window.java:2746)
2423   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.awt.Component.dispatchEvent(Component.java:4711)
2424   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
2425   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue.access$500(EventQueue.java:97)
2426   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$3.run(EventQueue.java:709)
2427   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$3.run(EventQueue.java:703)
2428   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.security.AccessController.doPrivileged(Native Method)
2429   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
2430   1181.430 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
2431   1181.431 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$4.run(EventQueue.java:731)
2432   1181.431 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue$4.run(EventQueue.java:729)
2433   1181.431 STDERR (SwingExceptionHandler.java:49) 	at java.security.AccessController.doPrivileged(Native Method)
2434   1181.431 STDERR (SwingExceptionHandler.java:49) 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
2435   1181.431 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
[... cut ...]
2441   1181.431 STDERR (SwingExceptionHandler.java:49) 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2442   1181.431 INFO   (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
2443   1181.431 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
2444   1183.729 USER   (SwingExceptionHandler.java:220) User requested sending bug report
2445   1183.731 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

dkingsley, 2017-06-20:
I can not duplicate this in the unstable branch.

dkingsley, 2017-06-20:
I can reproduce this error on Win 8.1 Java 8 update 131.

dkingsley, 2017-06-21:
The exception thrown is the same as that described in issue 331, please refer to that issue.

commit 015437d3 (2018-01-06): [resolves #368][resolves #331] User may scale freeform fins.
closed 2018-01-22 by wcravens