openrocket/openrocket #332: Suggestion: Make Pod Set and Booster configuration tabs consistent with other component types

neilweinstock, 2017-02-24:
The config dialog for Pod Sets and Boosters are arranged differently from other components.  All the old components start with a "General" tab, followed by other component-specific stuff, then "Override", then "Appearance", then "Comment".  

For some reason the Pod Sets and Boosters have "Override" first, and then "Parallel", then "Comment".  I haven't been able to come up with any real justification for this divergence.  I think "Parallel" should be renamed "General", because it has parameters that are similar to what's found in the "General" tab on other component types.  And then it should go all the way on the left, again for consistency but also because it's the tab likely to be edited most frequently and the leftmost tab is opened by default when the dialog is first opened.

teyrana, 2017-05-29:
@neilweinstock 
I've fixed this in PR #337.  Please take a look!

neilweinstock, 2017-06-01:
Thank you so much!  My Java dev environment is in disrepair at the moment, but as I'll give it a try as soon as I can get it back together (or if you want to shoot me a JAR file I'll try it immediately).

ChrisMickelson, 2017-06-05:
I've done some testing, seems to work excellent with one exception: Automatic positioning doesn't work for pods, but it still does work for boosters as before. For the pod panel, the check box is there, but when checked, nothing changes.

teyrana, 2017-06-09:
Ah, I should've known.  I'll put that next on my list, then.

Daniel Williams
equipoise@gmail.com
858-220-8134 (Mobile)
339-368-6425 (Skype)
LinkedIn <https://www.linkedin.com/in/williamsdm>
GitHub <https://github.com/teyrana>

neilweinstock, 2017-06-09:
[https://mailguru.s3.amazonaws.com/images/spark/qr.v2/935cf1c585724824a8e35714de5e0627.png]

Thank you!

Spark by Readdle<https://sparkmailapp.com>

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2017-06-11:
@neilweinstock updated. Both boostesr and pods can auto-adjust radial offsets now.

Not the UI update fixes are being fixed separately.  I want to hear about them, but I will probably fix them separately.

teyrana, 2017-06-11:
Unit tests *are* actually passing-- The config issue which trips up Travis-CI is fixed in PR #350.

teyrana, 2017-10-09:
@neilweinstock @wcravens Request that we close this issue as resolved.

neilweinstock, 2017-10-10:
If it's fixed I'm OK with it, but I haven't had a chance to test the latest unstable release.  I wish there were a way to get latest JAR files without installing a full dev environment, but I guess I'll have to revive mine.

teyrana, 2017-10-10:
There are a couple people requesting pre-built jars... is there a way to
generate them from a jenkins or PR  hook ?

Daniel Williams
equipoise@gmail.com
858-220-8134 (Mobile)
339-368-6425 (Skype)
LinkedIn <https://www.linkedin.com/in/williamsdm>
GitHub <https://github.com/teyrana>

wcravens, 2017-10-28:
Dan confirmed that this is fixed in Unstable

closed 2017-10-28 by wcravens