openrocket/openrocket #334: Clipboard - Copy and Paste

eduncan911, 2017-04-12:
There is some quirkiness happening when copying components of a rocket from one instance of OpenRocket to another instance of OpenRocket.

(macOS, but I head Windows users have the same issue so I think it's a general issue)

Most likely OpenRocket java isn't monitoring clipboard activity.

**Does not work**
- Open an instance of OpenRocket, open your rocket.
- Goto File -> New (this creates another instance of OPenRocket).
- Go back to original rocket, select components, CMD-C (or CTRL-C).
- Switch to new instance, CMD-P (or CTRL-P) for paste.  
Nothing happens.

A work around we have been doing is:
- Create a New rocket, SAVE first.
- Open an instance of OpenRocket, open your rocket you want to copy from.
- Select the components you want to copy, CMD-C.
- THEN, go to File -> Open Recent and select the New rocket you saved previously.
- This creates a 2nd instance.
Now you can paste.

To repeat, you have to Save and close the New one you are working on and start over with the steps above.

ChrisMickelson, 2017-04-12:
I followed these steps exactly. Works for me. Did you click on Sustainer first before pasting? You need to tell it where to paste.

eduncan911, 2017-04-14:
I see the issue.

Yes, we must click on Sustainer/component tree first and the paste works every time.  

I was clicking down into the 3D/design view.

Closing!

closed 2017-04-14 by eduncan911

plaa, 2017-04-14:
Note that if you start a new instance of OpenRocket (from command line or double-clicking the JAR) then copy-paste won't work between the two instances because the copy-paste is built in-app.  You need to open the new window or file from the existing instance.