openrocket/openrocket #340: Include instructions for building from source

tibbe, 2017-06-05:
It would be nice if the readme (or some similar file) included instructions on how to build the project from source, to ease contribution or for people who want to try the bleeding edge version.

tibbe, 2017-06-05:
I guess this mostly covers it: http://wiki.openrocket.info/Developer%27s_Guide

Perhaps instructions for how to actually build the jar would be helpful.

ChrisMickelson, 2017-06-05:
There may be several ways depending on your environment but what I do if I want to build a jar is (first download the source as zip from GitHub and extract to a folder). Then in my IDE of choice(IntelliJ IDEA 2016.3.4), I just open the project (File->Open->your project folder) folder and run the Ant build script which is the "build.xml" file in the root of the project folder. (View->Tool Windows->Ant Build), in the window to the right, click the green plus, navigate to your project folder, expand it and select the build.xml file, click OK. Then click the green triangle Run button. After a minute or so once it's finished, it will place an "OpenRocket.jar" in your project folder, something like \openrocket-unstable\swing\build\jar\OpenRocket.jar

tibbe, 2017-06-05:
Thanks! I managed to build the jar now. I'm experimenting with building on JDK9 so I can get HiDPI working and get legible fonts on my laptop screen.

closed 2017-06-05 by tibbe

teyrana, 2017-06-05:
+1 For the large screen love!  It definitely could use some help :)