openrocket/openrocket #344: LogBack instructions unclear

tibbe, 2017-06-06:
I tried to follow the instructions at

http://wiki.openrocket.info/Developer's_Guide#LogBack

to enable logging to stdout without any luck. In particular adding

```
-Dlogback.configurationFile=config/logback-stdout-level-error.xml 
```

to the "Program arguments" in the run configuration setup for SwingStartup doesn't seem to work. The above setting seems off, because I think eclipse runs the project using swing/, not core/, as the working directory and the config file is in the latter. Trivially changing the path to an absolute path had no effect.

tibbe, 2017-06-06:
I finally got something to the console by putting this in the *VM arguments*:

```
-Dlogback.configurationFile=/usr/local/google/home/tibell/src/openrocket/core/config/logback-stdout-level-error.xml
-Dopenrocket.log.stdout=debug
-Dopenrocket.debug
```

dkingsley, 2017-06-20:
Is this an issue with the code or documentation for debugging?

tibbe, 2017-06-20:
I would lean towards documentation.

neilweinstock, 2021-09-13:
Fixed by @SiboVG, closing.

closed 2021-09-13 by neilweinstock