openrocket/openrocket #346: Starting SwingStartup directly works, but not OpenRocket

tibbe, 2017-06-07:
I can launch `SwingStartup` under Java 9 but using the `OpenRocket` entry point (which OpenRocket.jar uses) leads to the below exception.

I suspect `JarInJarStarter` is to blame somehow. Perhaps it interacts badly with the class path/module mechanisms of Java 9, given that it reconstructs the class path.

Possible solution: Is it possible to pass `--add-modules=java.se.ee` to the class loader that invokes `SwingStartup` from `OpenRocket`? That is a commonly recommended workaround for this particular `NoClassDefFoundError` exception. The issue is that Java 9 doesn't expose that API by default anymore (it was moved to a different module).

(Whether we can land Java 9 support in "unstable" yet is a different question. Right now I'm just trying to build a working jar locally.)

(The `--permit-illegal-access` is expected and unrelated.)

```
WARNING: --permit-illegal-access will be removed in the next major release
New classpath:
   file:/usr/local/google/home/tibell/src/openrocket/swing/bin/
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/jogl/jogl-all.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/iText-5.0.2.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/jcommon-1.0.18.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/jfreechart-1.0.15.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/OrangeExtensions-1.2.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/jogl/gluegen-rt.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/bin/
   file:/usr/local/google/home/tibell/src/openrocket/core/lib-extra/RXTXcomm.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/resources/
   file:/usr/local/google/home/tibell/src/openrocket/core/lib/opencsv-2.3.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/lib/javax.inject.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/lib/aopalliance.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/lib/slf4j-api-1.7.5.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/lib/annotation-detector-3.0.2.jar
   file:/usr/local/google/home/tibell/src/openrocket/lib-test/hamcrest-core-1.3.0RC1.jar
   file:/usr/local/google/home/tibell/src/openrocket/lib-test/hamcrest-library-1.3.0RC1.jar
   file:/usr/local/google/home/tibell/src/openrocket/lib-test/jmock-2.6.0-RC2.jar
   file:/usr/local/google/home/tibell/src/openrocket/lib-test/jmock-junit4-2.6.0-RC2.jar
   file:/usr/local/google/home/tibell/src/openrocket/lib-test/junit-dep-4.8.2.jar
   file:/usr/local/google/home/tibell/src/openrocket/lib-test/test-plugin.jar
   file:/usr/local/google/home/tibell/src/openrocket/lib-test/uispec4j-2.3-jdk16.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/lib/guice-4.1.0.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/lib/guice-multibindings-4.1.0.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/logback-classic-1.0.12.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/logback-core-1.0.12.jar
   file:/usr/local/google/home/tibell/src/openrocket/core/lib/guava-20.0.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/resources/
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/miglayout-4.0-swing.jar
   file:/usr/local/google/home/tibell/src/openrocket/swing/lib/rsyntaxtextarea-2.5.6.jar
115            INFO  [main] n.s.o.s.SwingStartup - Starting up OpenRocket version 15.03dev
122            INFO  [main] n.s.o.s.SwingStartup - Checking for graphics head
125            INFO  [main] n.s.o.s.SwingStartup - Moving startup to EDT
286            INFO  [AWT-EventQueue-0] n.s.o.s.SwingStartup - Initializing the splash screen
288            INFO  [AWT-EventQueue-0] n.s.o.s.SwingStartup - Registering exception handler
295            INFO  [AWT-EventQueue-0] n.s.o.s.SwingStartup - Loading databases
WARNING: Illegal access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/usr/local/google/home/tibell/src/openrocket/core/lib/guice-4.1.0.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain) (permitted by --permit-illegal-access)
1217           ERROR [main] n.s.o.g.m.SwingExceptionHandler - Handling uncaught exception on thread=Thread[main,5,main]
java.lang.RuntimeException: Error starting OpenRocket
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40) ~[bin/:na]
	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:36) ~[bin/:na]
Caused by: java.lang.reflect.InvocationTargetException: null
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[na:na]
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[na:na]
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[na:na]
	at java.base/java.lang.reflect.Method.invoke(Method.java:563) ~[na:na]
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38) ~[bin/:na]
	... 1 common frames omitted
Caused by: java.lang.reflect.InvocationTargetException: null
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1323) ~[na:na]
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1298) ~[na:na]
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1475) ~[na:na]
	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:76) ~[bin/:na]
	... 6 common frames omitted
com.google.common.util.concurrent.ExecutionError: com.google.common.util.concurrent.ExecutionError: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.common.cache.LocalCache$Segment.get(LocalCache.java:2212) ~[guava-20.0.jar:na]
[... 62 lines cut ...]
	at com.google.inject.internal.InjectorImpl.getInternalDependencies(InjectorImpl.java:616) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.InjectorImpl.cleanup(InjectorImpl.java:572) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:558) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:887) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:808) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:285) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:217) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.SingleFieldInjector.<init>(SingleFieldInjector.java:42) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.MembersInjectorStore.getInjectors(MembersInjectorStore.java:131) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.MembersInjectorStore.createWithListeners(MembersInjectorStore.java:98) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.MembersInjectorStore.access$000(MembersInjectorStore.java:37) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:45) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:42) ~[guice-4.1.0.jar:na]
	at com.google.inject.internal.FailableCache$1.load(FailableCache.java:37) ~[guice-4.1.0.jar:na]
	at com.google.common.cache.LocalCache$LoadingValueReference.loadFuture(LocalCache.java:3628) ~[guava-20.0.jar:na]
	at com.google.common.cache.LocalCache$Segment.loadSync(LocalCache.java:2336) ~[guava-20.0.jar:na]
	at com.google.common.cache.LocalCache$Segment.lockedGetOrLoad(LocalCache.java:2295) ~[guava-20.0.jar:na]
	at com.google.common.cache.LocalCache$Segment.get(LocalCache.java:2208) ~[guava-20.0.jar:na]
	... 52 common frames omitted
Caused by: java.lang.ClassNotFoundException: javax.script.ScriptEngineFactory
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:466) ~[na:na]
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:553) ~[na:na]
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:486) ~[na:na]
	... 76 common frames omitted
```

tibbe, 2017-06-07:
Here's an example of the "standard" solution to javax.script not being exposed by default anymore: https://stackoverflow.com/questions/43574426/how-to-resolve-java-lang-noclassdeffounderror-javax-xml-bind-jaxbexception-in-j

tibbe, 2017-06-18:
The issue seems to be that the class loader is created without a parent class loader. If we change

``` java
ClassLoader loader = new URLClassLoader(urlArray, null);
```

in swing/src/net/sf/openrocket/startup/jij/JarInJarStarter.java to

``` java
ClassLoader loader = new URLClassLoader(urlArray);
```

the application launches successfully. I'm not 100% sure why this is. It could have something to do with the new module layers in Java 9, but I'm not sure.

teyrana, 2018-01-13:
@tibbe @wcravens 

Hey, bad news-- I discovered that this fix interferes with building from ant and the command line. Any ideas? 

### Build:
`ant jar` 
### Run
`java -jar swing/build/jar/Openrocket-18.01-dev.jar`

### Java Version
``` bash
13:26::%> java -version 
openjdk version "1.8.0_151"
OpenJDK Runtime Environment (build 1.8.0_151-8u151-b12-0ubuntu0.17.04.2-b12)
OpenJDK 64-Bit Server VM (build 25.151-b12, mixed mode)
```

teyrana, 2018-01-13:
FYI, I'll be removing this from rc4 until we get this resolved.
concerns: 
- the project's included ant `build.xml` should be useful, or fixed to be useful.
- how do we build new release jars? 

Offhand, reasonable acceptance criteria should be that the patch compiles & runs cleanly from ant, eclipse and intellij.  Thoughts?

wcravens, 2018-01-13:
The release instructions are in core/release.txt.  But they need updated a
little.

Build.xml should be correct, and I agree with your acceptance criteria.

I have the access that I need to update the 'update.php' system.  I was
hoping to get it switched over to a REST API that doesn't run on
SourceForge, but if a release is happening really soon I may just stick
with what we have.

Wes

closed 2018-01-22 by wcravens

Suhail, 2018-06-19:
I got here because when I ran open rocket for the first time, it totally crashed & I had a hard time getting it up and running on my mac. I am using pretty up-to-date software: Mac OS 10.13 & JDK 10. If you're here because you can't get this running, here's the solution:

1. OpenRocket currently works on JDK 8 and if you're on something better than that, it's not going to work. That means you have to run it on an older JDK (namely JDK 8)

2. Install JDK 8 (I downloaded jdk1.8.0_171)

3. Then, run the command: /usr/libexec/java_home -v 1.8.0_171 --exec java -jar ~/Downloads/OpenRocket-15.03.jar (This will run OpenRocket using JDK 8 on your Mac even if you have a newer version)

Make sure your paths are correct in step 3.

teyrana, 2018-06-21:
Aaaaactually,  that's a really good synopsis of the situation !

Thanks :D

Daniel Williams
equipoise@gmail.com
858-220-8134 (Mobile)
github.com/teyrana
<https://github.com/teyrana>LinkedIn
<https://www.linkedin.com/in/williamsdm>

wcravens, 2018-06-21:
Just so that everyone is aware...  this has always been a typical situation for Java applications.  The only alternative is to package a JVM with your app, which makes for a very large download.  However this isn’t as much of a barrier these days with modern bandwidth.

We’ll be considering options for solutions to this issue after the next release.

Wes

PR #384 (2018-01-06): Merge RC3 into unstable (current-version 18.01-rc1)
Resolves: #346 #369 #377 #379 #368 #331 #374 #371 #367 #328 #323 

This is another batch of bug-fixes, on the way to a releasable version.