openrocket/openrocket #347: Make work with Java 9

tibbe, 2017-06-07:
This is an umbrella issue for making OpenRocket work with Java 9, once it's released. Supporting Java 9 has several benefits, perhaps foremost making OpenRocket work on HiDPI screens.

There are several issues to resolve e.g. #346. The major ones relates to how we build the final jar (with bundled deps) and how we make sure the right modules (e.g. `java.scripting`, which contains `javax.script.ScriptEngineFactory`) are exposed to the loaded code (since we use a class loader).

teyrana, 2018-02-11:
haha, this should be relevant: 
https://docs.oracle.com/javase/9/migrate/toc.htm#JSMIG-GUID-D867DCCC-CEB5-4AFA-9D11-9C62B7A3FAB1

> The changes in JDK 9 may impact code that creates class loaders with null (that is, the bootstrap class loader) as the parent class loader and assumes that all platform classes are visible to the parent. Such code may need to be changed to use the platform class loader as the parent (see ClassLoader.getPlatformClassLoader).

teyrana, 2018-02-11:
Probably solution: 

[ Multi-Release Jars ]( http://openjdk.java.net/jeps/238 )

tibbe, 2018-05-18:
This is what the failure looks like:

```
java --version
java 10.0.1 2018-04-17
Java(TM) SE Runtime Environment 18.3 (build 10.0.1+10)
Java HotSpot(TM) 64-Bit Server VM 18.3 (build 10.0.1+10, mixed mode)
```

```
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.RuntimeException: Error starting OpenRocket
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40)
	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:32)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38)
	... 1 more
Caused by: java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(Unknown Source)
	at java.desktop/java.awt.EventQueue.invokeAndWait(Unknown Source)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(Unknown Source)
	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:76)
	... 6 more
Caused by: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.ConstructorInjectorStore.get(ConstructorInjectorStore.java:49)
	at com.google.inject.internal.ConstructorBindingImpl.initialize(ConstructorBindingImpl.java:125)
	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:521)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
	at com.google.inject.internal.FactoryProxy.notify(FactoryProxy.java:46)
	at com.google.inject.internal.ProcessedBindingData.runCreationListeners(ProcessedBindingData.java:50)
	at com.google.inject.internal.InternalInjectorCreator.initializeStatically(InternalInjectorCreator.java:133)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:106)
	at com.google.inject.Guice.createInjector(Guice.java:95)
	at com.google.inject.Guice.createInjector(Guice.java:72)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at net.sf.openrocket.startup.SwingStartup.runInEDT(SwingStartup.java:143)
	at net.sf.openrocket.startup.SwingStartup.access$000(SwingStartup.java:43)
	at net.sf.openrocket.startup.SwingStartup$1.run(SwingStartup.java:79)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue.access$600(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
[... 248 lines cut ...]
  84      0.895 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
  85      0.895 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
  86      0.895 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
  87      0.895 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
  88      0.895 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
  89      0.895 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.SingleFieldInjector.<init>(SingleFieldInjector.java:41)
[... cut ...]
 101      0.896 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
 102      0.896 STDERR (SwingExceptionHandler.java:49) 	... 67 more
 103      0.896 INFO   (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
 104      0.898 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 105      5.176 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 106      5.179 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------
```

kernelsmith, 2019-03-14:
I'm also erroring out while trying to start 15.03 on Linux (Open JDK 11.0.3) and MacOS (Oracle Java 11.0.2). I'm of course assuming this is because Open Rocket only supports 8, a full 3 major versions behind. If we're unlikely to be current-version-compatible anytime soon, we should at least mention the compatibility on the main page and certainly on the packages page http://openrocket.info/packages.html This is my second time trying to get OpenRocket to run and I wasted a good 30 mins trying to track down the stack trace, upgrading packages etc. I really could have benefitted from the simple knowledge about compatibility.

lectroidmarc, 2019-04-19:
Yeah, I just ran into this on Ubuntu 19.04.  I used to have java-8 installed under 18.10, and was able to run OpenRocket, but it appears there's no java-8 for Ubuntu 19.04 anymore.

neilbalch, 2019-10-27:
I too need this software to run on >Java8. 😦 If there's any development to be done in order to support higher Java versions, I'd be happy to volunteer my time.

tarnh, 2019-10-27:
@neilbalch Just do it ;)

teyrana, 2020-02-13:
Ongoing work at #556

teyrana, 2020-03-13:
Probably fixed by #572.  (Fix is now in mainline development)

itshanney, 2020-04-06:
@teyrana propose to close this as mainline development is now on Java 9+.

closed 2020-04-06 by teyrana