openrocket/openrocket #355: Travis-CI automatic tests are failing for all code

teyrana, 2017-06-24:
This is due to a configuration change on the Travis servers, which prevents our old build files from running correctly.  Fix submitted in PR #350.

teyrana, 2017-06-24:
Output from one of the failed builds:  This particular build is not important, this same error happens on all of our CI test builds, now....
```
BUILD FAILED
/home/travis/build/openrocket/openrocket/build.xml:49: The following error occurred while executing this line:
/home/travis/build/openrocket/openrocket/core/build.xml:95: Problem: failed to create task or type junit
Cause: the class org.apache.tools.ant.taskdefs.optional.junit.JUnitTask was not found.
        This looks like one of Ant's optional components.
Action: Check that the appropriate optional JAR exists in
        -/usr/share/ant/lib
        -/home/travis/.ant/lib
        -a directory added on the command line with the -lib argument
Do not panic, this is a common problem.
The commonest cause is a missing JAR.
This is not a bug; it is a configuration problem
```

[Full output can be found here](https://travis-ci.org/openrocket/openrocket/builds/246631105?utm_source=github_status&utm_medium=notification)

ddekany, 2017-08-05:
I have ran into this problem today too. I guess `ant-optional` (Debian package name) is not installed anymore,  and for some reason this change has reached us (https://travis-ci.org/apache/incubator-freemarker) only in recent days. As temporary workaround, I have added this to `.travis.yml`:

```
before_install:
- sudo apt-get -qq update
# ant-optional is needed for ant junit
- sudo apt-get install ant-optional
```

Obviously, it's pointless to have Ant preinstalled without JUnit support on a CI server. Almos all projects that care to use CI will use JUnit as well. Fix this please, this issue is open for more than a month now!

teyrana, 2017-08-05:
@ddekany  I don't follow -- there is already a working patch for this (#350), just waiting for merging.

Do you think installing ant-optional would be a better solution?  I'm not familiar with ant, so it may well be.  Please let me know what you think.

ddekany, 2017-08-06:
@teyrana I have reported this Travis CI bug to the wrong tracker, but then has left it here as it may helps here. The intended report is here: https://github.com/travis-ci/travis-ci/issues/7706 
Otherwise, yes, I think since the issue is caused by incomplete Ant installation in the Travis VM-s, the proper fix is to fix that.

SpamWall, 2017-08-06:
Have any changed been merged with unstable recently?  It seems like there are quite a few pull requests waiting to be merged.  Having such a log delay means other changes will start to have conflicts that make merging more difficult.  Can we set some target period for PRs to either be rejected or merged?

 

Craig

PR #365 (2017-09-28): Bugfix Aggregation: release-candidate-0
Fixes numerous bugs -- See individual commits for details.

Fixes: #355 #332 #330  ... and other issues with out bug-report-numbers.
closed 2017-10-09 by teyrana