openrocket/openrocket #357: Transition Templates not all showing up on report

fulldec, 2017-06-29:
I cannot get OR to show all transition templates in the Design Report. I am designing an egg capsule composed of 4 cardstock transition elements, nose,  2 body tube sections, and transition to the BT-20. The Body Tube sections are cylindrical transitions. Those are the ones that do not show up (sometimes 1 shows up, but not both). Through the design iterations, sometimes 3 of the 4 transition templates show up, sometimes only 2. 

User MikeyDSlagle  on TRF changed the "aft diameter" on the smaller one from automatic to 1.811" and it works now.

Am using 15.03 on a Mac. 

Attached is a typical problem file with extension changed to TXT

[Eggloft cardstock B Alt rev 3.txt](https://github.com/openrocket/openrocket/files/1113146/Eggloft.cardstock.B.Alt.rev.3.txt)
.

dkingsley, 2017-06-29:
I tested this on both 15.03 and the unstable branch and using the supplied file and can reproduce the error.

hcraigmiller, 2022-08-30:
This issue can no longer be replicated, suggest confirmation and closure.

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

JoePfeiffer, 2022-09-03:
Concur.

closed 2022-09-03 by JoePfeiffer