openrocket/openrocket #360: Invert stage numbering

Symbian9, 2017-07-30:
There are two ways for numbering stages in multistage rocket

* 1st stage - upper stage of rocket

* 1st stage - lower stage of rocket

In OpenRocket there is only one way for numbering - "1st stage is upper"

So, add setting "Invert stages numbering", because in most cases in traditionaly "1st stage is lower stage".

# REFERENCES

* https://en.wikipedia.org/wiki/Multistage_rocket
* http://steamcommunity.com/app/220200/discussions/0/648814842625230911/
* https://www.reddit.com/r/KerbalSpaceProgram/comments/475glk/when_will_squad_fix_the_stage_numbers/

dkingsley, 2017-08-10:
Please feel free to implement this feature and create a pull request.

plaa, 2017-08-10:
I think this warrants more design discussion before implementation / integration.  There's many ways to implement this, each with downsides.

Personally I'm against any unnecessary settings, as they are a source of complexity and confusion.  First discussion is can we agree on one numbering.  If a setting is required, is it per-rocket or per-installation?  Both are a potential source of confusion, e.g. when discussing designs.

teyrana, 2018-06-16:
## Definitions: 
(for brevity, I suggest the following names) 

"top-down" == upper stage first
"bottom-up" == lower stage first
"ignition order" == in order of first ignition (99% identical to 'bottom-up', but may differ in some corner cases)

## Discussion

1. Which is more popular? 

I was taught the ignition order version as the default in engineering classes, and in NAR discussions. 
Thoughts? 

2. How do we order pods & boosters? 

Suppose your lower stage has strap-on boosters that are aligned with the centerline/core stage. Which is the first stage? 

Suppose they are both ground-ignition (i.e. at t=0).   Does this change your answer? 

Compare: 
- Long March / CZ-2F (https://cdn.britannica.com/668x448/88/159588-004-37F229A0.jpg)
- GSLV Mk3 (https://www.livemint.com/rf/Image-621x414/LiveMint/Period2/2017/03/18/Photos/Processed/gslv-k1g--621x414@LiveMint.jpg)

## Implementation:
Under the hood, I suppose we'd just add unique, random ids to each stage, and then the UI-order is just for human consumption.   We already used GUIDs elsewhere, and it's probably the easiest addition.

wcravens, 2019-02-05:
Closing per thread on the email list.

PR #467 (2018-11-03): Show stage names instead of stage numbers
Close #360 and #460 by showing stage names instead of numbers
closed 2019-02-05 by wcravens