openrocket/openrocket #361: 2nd stage not showing in simulation

DyingEcho, 2017-08-14:
I'm new to rocketry so I may be making an error myself, but I've double-checked everything I can think of.

I've designed a 3-stage rocket with parachutes in the transition parts of each stage. It looks like this:
[image]

I've set up the recovery system like this:
[image]

And the simulation settings are optimal launch conditions.
[image]

However, when I simulate and plot the side profile of the launch, the menu allowing me to select only one stage to view only shows 2 stages - Launcher (3rd) and Sustainer (1st). What's happening to the second stage?

Also, it appears that the Sustainer's firing is ineffective as it doesn't go much further than the Launcher. The separation also occurs long before the ejection charges are fired.
[image]

Have I done something wrong or is this a problem with OpenRocket? 
Thanks, DE

wcravens, 2017-08-14:
Can you attache the ORK file?  There are setting that adjust when the different stages ignite.  It could be something wrong in those.

Thanks,

Wes

DyingEcho, 2017-08-14:
[main.ork.zip](https://github.com/openrocket/openrocket/files/1222336/main.ork.zip)
Here it is

ChrisMickelson, 2017-08-16:
No fins on the other stages might be a problem. When the first stage drops off, no more guidance. Rocket hits the ground before the next stage and no more plotting. Are you getting "Large angle of attack" warning? Each stage needs some sort of fins to keep it straight.

DyingEcho, 2017-08-16:
I am getting that warning, I'll add fins and see what happens. Thanks!

DyingEcho, 2017-08-16:
I've changed to this configuration:
[image]
But I'm still getting something similar:
[image]

I'm no longer getting 'large angle of attack' but I'm being told that recovery devices were deployed at a high speed while motors were still burning.

JoePfeiffer, 2018-05-31:
I'll take a look at this if no one has any objection.

JoePfeiffer, 2018-09-29:
I hope you haven't thought I'd gone away and forgotten this!  There's a lot of debugging going on trying to get the next release ready to go, and every time I looked at your issue I found a staging-related but in the new version.  Anyway, the basic problem you were seeing is you were testing a really heavy rocket with a really small motor, and so you didn't have enough velocity coming off the rail to be stable.  When I modified the design to use a polystyrene nose cone, cardboard body buts and transitions, and basswood fins, and used G64 motors, I got reasonable flights.  It still getting exceptions when the first and second stage ejection charges are going off, which looks like a bug in the simulation -- so I'll be seeing if the development version is doing it too...

teyrana, 2018-09-29:
Hey, we have a couple pull requests piling up...

Who has merge permissions these days?

wcravens, 2018-09-29:
That would be at least me.  I've been wondering about a strategy for
managing pull request merges.  I'm carefree enough to merge anything that
is passing tests.  But I don't have the time to build/run the app and test
things.  I'm happy to just merge the pull requests as you all provide them,
but I wouldn't mind a little more sanity checking before we do so.  For now
I'll just merge the latest but going forward could we figure out a way to
verify pull requests before I merge them?  I looked into Github's
multi-reviewer strategy
<https://blog.github.com/2018-03-23-require-multiple-reviewers/>but didn't
get very far.

Wes

wcravens, 2018-09-29:
I should also add that I won't merge anything with a conflict.  If there is
a merge conflict it should be corrected and then resubmitted.

Wes

JoePfeiffer, 2018-09-29:
All I know for sure is it isn't me!  I don't really know what a good policy would be, especially...  not merging anything with conflicts seems like a no-brainer.  Thanks for that batch of commits.

teyrana, 2018-09-29:
Hey, I would *love* to get secomd opinions on my stuff... It would save us
from some of my more hair-brained mistakes.

However, I will point out, we don't exactly have a glut of people looking
to contribute here...

I'm up for trying it, though!

JoePfeiffer, 2019-02-22:
This turns out to not be a bug in the simulation.  What's happening is that (1) at liftoff, the entire rocket is being simulated as a single entity.  This is simulation branch "Sustainer".  (2) at motor burnout, stage separation occurs; we now have simulation branches "Sustainer" (simulating the Sustainer and the Booster Stage) and "Launcher".  (3) Booster Stage's motor ignition occurs at the end of the delay charge for the Launcher. However, both parts land before this event occurs.  Each simulation branch is terminated when the part of the rocket it is simulating lands, so we get the reported symptom:  it appears as if only the Launcher and Sustainer are simulated.

It does seem as if we want to do a better job showing the user what's going on.  My thought on the best way to do this is to run each simulation branch to the bitter end, processing all of its flight events even after each part of the rocket has landed.  I submitted PR  521 a couple of minutes ago to do this.  Here's a plot of the simulation of this rocket, with the PR applied:
[image]
and here's the warnings that got recorded.  I deliberately let warnings appear more than once, in part because they are actually new warnings from different simulation branches, and in part to better show the seriousness of the situation.
[image]

teyrana, 2020-02-15:
@iDoObject @JoePfeiffer 

Fix merged into 'unstable' in Pull Request #521. 

Does this resolve this issue?

iDoObject, 2020-02-15:
Wow! I ignored GitHub for a while (if 4 months counts as 'a while') and came back to this. Thanks for putting so much effort into this random issue I filed in 2017 from a poor understanding of rocket design 😄

That should resolve things. Thanks again!

teyrana, 2020-02-15:
Awesome !

Thanks for checking in :)

closed 2020-02-15 by teyrana