openrocket/openrocket #362: Problem scripting with OpenRocket!

EngMaronas, 2017-08-28:
Hello fellow rocket scientists,

I need to use OpenRocket scripting and I'm trying to use the guide from OpenRocket wiki, but it seems to be out of date.Which configurations/version of OpenRocket I need to run .py scripts?

I have tried many versions of it, and in all of them I'm having a diferent problem.The closer I got from running a code was when I started to realize some fuctions used in orhelper.py are private in the OpenRocket 12.03, but in OpenRocket 15.XX they dont exist anymore.The functions like "startup.initializeLogging()" are private in older OpenRocket, but doesnt exist in the new...

There are any OpenRocket version where orhelper.py and scripts works out of the box?

What do i need to run/script py codes to OpenRocket?

Thanks in advance for this amazing open source software,

Maronas.

jonasdegrave, 2018-10-12:
Up! This is really important, does anybody have a workaround or suggestions for this yet?

`~ $ python simple_plot.py`

`
Traceback (most recent call last):
  File "simple_plot.py", line 6, in <module>
    with orhelper.OpenRocketInstance('OpenRocket-test.jar'):
  File "/home/.../or-jpype-examples/orhelper.py", line 19, in __init__
    orp.startup.Startup.initializeLogging()
  File "/usr/local/lib/python2.7/dist-packages/jpype/_jpackage.py", line 62, in __call__
    raise TypeError("Package {0} is not Callable".format(self.__name))
TypeError: Package net.sf.openrocket.startup.Startup.initializeLogging is not Callable
`

yuriachermann, 2019-03-17:
@EngMaronas @jonasdegrave *If you can work this out, please let me know. I would really appreciate!*

SiboVG, 2022-02-11:
Using the latest unstable build, I was able to run `simply_plot.py` without any problems. But I would like this to be tested some more after the release comes out before we close this issue.

hcraigmiller, 2022-04-03:
See also, Issue #308.

SilentSys, 2022-04-03:
Can this issue be closed now seeing as the wiki has been updated with a working solution?

https://github.com/openrocket/openrocket/wiki/Scripting-with-Python-and-JPype
http://wiki.openrocket.info/Scripting_with_Python_and_JPype#Scripting_openrocket_with_Python_and_JPype
https://github.com/SilentSys/orhelper

closed 2022-04-03 by SiboVG