openrocket/openrocket #372: Suggestion: distribute packaged versions of the JAR file

neilweinstock, 2017-10-19:
Packaging the jar file as a self-contained app solves many problems for the user, and is also now the recommended method of distribution as per Oracle (http://www.oracle.com/technetwork/java/javase/eol-135779.html).  Suggest distributing packaged versions for Mac and Windows (maybe Linux too, although I haven't played with that).

When I created packaged versions in the past, I used the official [Oracle javapackager](https://docs.oracle.com/javase/8/docs/technotes/guides/deploy/packager.html) tool.  Instructions are provided for Mac; Windows is a bit of a pain because it requires an additional 3rd party tool.

Someone here at my company used Launch4j to package a JAR file for Windows.

Googling around I found [Packr ](https://github.com/libgdx/packr) on Github.  This might be best of all because it seems to offer a similar command line interface for Mac, Windows, and Linux.  I haven't tried it yet.

itshanney, 2020-04-06:
@neilweinstock assigning to me. Can you try out the Windows or Mac installers linked below?

- Windows: https://www.dropbox.com/s/ohiznsdzfllilxn/OpenRocket_windows-x64_20_x.exe?dl=0
- macOS: https://www.dropbox.com/s/2p8hf7b85wix1sm/OpenRocket_macos_20_x.dmg?dl=0

neilweinstock, 2020-04-07:
I sent feedback on the dev mailing list regarding the Mac installer, here it is again:
There are some aspects of it that I'm not too crazy about, although not necessarily dealbreakers:
1) the installer is actually itself an app.  That is pretty non-standard on the Mac.  Normally it'd either be an installer bundle or an app ready to be copied to /Applications.
2) It puts the app inside a folder inside /Applications, alongside the jar file.  I don't know why the jar file isn't inside the app bundle.
3) Also in the folder is an uninstaller app.  That is very un Mac-like.  Normally to delete an app you just drag the app to the Trash.

The icon needs work.  It seems to only be there at low resolution, needs to have higher-res versions as well.  That's easy to fix.  It also suffers from being unsigned at this point, although can still run in the usual way by going into security preferences and asking it to run.  Would be great if we could eventually get it signed, not sure what's involved in that (other than needing a developer account, which I and I'm guessing some other folks have).  That is something for the future, certainly.
I'll try the Windows installer and provide feedback tomorrow.

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

neilweinstock, 2020-04-07:
Hey I just noticed that the packaged Mac app does not put up the splash screen when it starts up.

teyrana, 2020-04-07:
Test report: 
--- 
Platform: Windows 10

1. First impression 
When you click on it, you get a giant, red warning splash screen.    Presumably that's the signing thing.  (yea, we'll want to fix that)

2. How to find
Then it proceeds easily enough.   It's available from the start menu, although it doesn't install anything to the desktop-- which I'm fine with.

3. Just Works
Clicking on the installed icon works as expected.  (no splash screen, here, either... but I've stopped noticing when we expect that, or not....)  It loads in under a second, so I don't see a need for one, anyway.

4. Uninstall
The uninstaller is available via the expected ways: "settings" > "Apps" > search for "openr" > click "uninstall"  It's definitely bare bones, but it works as expected.

neilweinstock, 2020-04-07:
I just tried the Windows installer, similar experience as Daniel.  Seems fine.

It does *not* seem to associate the new version with .ORK files, though; those are still opening with my old packaged version.  Did you include file associations in these installers?

itshanney, 2020-04-18:
@neilweinstock the latest installer bundle below should associate with .ork files. Can you confirm?

https://www.dropbox.com/s/2p8hf7b85wix1sm/OpenRocket_macos_20_x.dmg?dl=0

hcraigmiller, 2020-12-14:
Speaking only to windows, after using the installer, OpenRocket is run by executing "openrocket.exe." As part of the installation, if the file extension .ork was associated with openrocket.exe, then a double-click on an .ork file would open that file in OpenRocket and each .ork file would have the OpenRocket logo in front of the file name. Like this:

[image]

Otherwise you have to manually associate any .ork file with openrocket.exe, which will associate all .ork with this executable file.

neilweinstock, 2022-02-25:
With great joy I am closing this one.

closed 2022-02-25 by neilweinstock