openrocket/openrocket #375: Suggestion: for pods, allow automatic position and radio offset to be used together

neilweinstock, 2017-10-27:
My long treatise on how pod positioning could or should work will wait for another day, but in the meantime there is a _very tiny_ enhancement that would make the current implementation much more useful.

Right now, when "automatic positioning" is selected, the pods are positioned so the largest BT in the pod will land flush with the largest BT in the parent sustainer (this is sometimes problematic and a good subject for a separate issue).  The Radial Offset is greyed out and disabled.

Instead, when automatic position is selected, I propose to use Radial Offset to represent an additional positive _or negative_  offset to the automatically calculated position.

This would be so incredibly useful because in many cases, I want my pods to be a particular distance from the surface of the core BT, let's say 1".  As it stands now, if I want to do this I have to pull out the calculator: Radial Distance = 1/2*(coreBTDiameter + podDiameter)  + 1".  With my proposed feature I just need to check "automatic positioning" and enter 1" into radial offset and I'm done, even if BT diameters change.

As far as I can figure, this would be very simple to implement and would immediately make things work much better.

ChrisMickelson, 2017-10-28:
I checked in RC1, and if you use automatic positioning, close the dialog, then go back in, I'm seeing the calculated value filled in although still greyed out. Then if you un-check automatic positioning, it's no longer greyed out and you can and click the up arrow for an extra inch to what the automatically calculated value was.

neilweinstock, 2017-10-29:
That is very good to know, and in the absence of implementation of my suggestion I will take advantage of that.  However, the above suggested implementation is still greatly superior, for multiple reasons, but first and foremost because it will remember and retain the relationship as the design is modified.

teyrana, 2017-10-29:
It may be noted that this behavior is not present  On any other
components... Thus it wont be a trivial feature.

neilweinstock, 2017-10-30:
I can understand if this can’t make it into the current release, but it would be very useful.

- Neil

—

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

closed 2018-01-28 by wcravens