openrocket/openrocket #376: Suggestion: use rearmost sustainer BT to calculate automatic pod positioning

neilweinstock, 2017-10-27:
This one might be contentious but here goes.  It would be good to fix this before the release goes out, so a future change doesn't break existing designs.

Right now, automatic pod positioning uses the largest BT in the parent sustainer for its calculations.  I believe this is the wrong choice for the majority of designs.  Instead, the rearmost BT is the one where pods are likely to be mounted.  Consider any design where there is an enlarged payload section in the front: in these cases the diameter of the payload section BT has absolutely nothing to do with the mounting of pods in the back.

Using the rearmost BT should either be the only choice, or else the default in case it's made user-selectable.

There will be times when using the largest BT in the sustainer is actually what I want, in which case I could easily work around it by putting a right-sized phantom BT in the back of the sustainer, or else just use an additional radial offset on the pod positions (see [issue #375](https://github.com/openrocket/openrocket/issues/375)).  In the current implementation, though, if I want pods flush with a smaller rear BT I have no choice but to abandon automatic positioning.

A more elaborate solution would be to allow the user to choose the method, but this could be considered for the future and would not be necessary on day one, since perfectly acceptable workarounds would exist in the short term.

teyrana, 2017-12-31:
@neilweinstock Alternate solution idea:  We could make Parallel Boosters and Pods addable *only* to BodyTubes, which would be both more precise, and neatly solve this concern.   

Thoughts ?

neilweinstock, 2017-12-31:
Interesting thought... let me ponder a bit. Want to think through the implications.

- Neil

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

neilweinstock, 2018-01-02:
OK, my New Year’s haze is gone and I’m back to being semi-functional. ☺

Yes, add pods to body tubes, absolutely.  When I first looked at the builds of the new release I wondered why pods were only attached to the sustainer (or booster I suppose).  I assumed that there was a technical reason of some sort.  If in fact you can move them to the body tubes that would be a definite step in the right direction.

-          Neil

closed 2018-01-28 by wcravens