openrocket/openrocket #381: Difficulty converting RockSim files to OpenRocket format

woferry, 2017-12-12:
From a recent TRF discussion, at least on the Mac it seems that converting a .rkt file to .ork isn't working as expected.

So I open any .rkt file.  I add some motors and simulate them.  Now I go to File and Save as…  There is a pop-up in the dialog for File Format, but if you select OpenRocket designs it immediately snaps back to RockSim designs, it's impossible to leave OpenRocket selected.  If I change the file extension from .rkt to .ork, it saves the file as .ork.rkt (though it is much smaller than the original file, it is not a zip archive as expected for .ork, and the motor configurations added are not saved).

The only way to do the conversion I've found is to first rename the original .rkt file to have a .ork extension.  Now if you open it in OpenRocket you don't even have to Save as… to create a new file, as soon as you Save the file is saved in OpenRocket format (zip) instead of the RockSim format (ASCII) it was when it was opened.  So there is a workaround at least.

Seen in 15.03 on multiple macOS/java versions, I'm currently running 10.12.6 (16G1036) and java -version reports:
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468-11M4833)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)

wcravens, 2017-12-14:
Please provide an rkt file by attaching to this issue.

teyrana, 2017-12-30:
We also had quite a discussion on this since 15.03 was released, so it may have been fixed.  If you provide the file that caused this behavior, we can test it on the latest branch.

sjh114961, 2022-03-10:
This issue seems to be fixed in 22.02.beta.01 for windows,

closed 2022-04-03 by SiboVG