openrocket/openrocket #382: Debian Installer Fails to Move .jar File to Install Directory

Joedang, 2017-12-13:
When installing with `apt`, it ends with 

```
Downloaded: 2 files, 21M in 46s (471 KB/s)
removing old .jar files from /usr/lib/openrocket ...
moving download to /usr/lib/openrocket ...
mv: cannot stat 'download': No such file or directory
```

I don't know anything about Debian-style installers, but I'm guessing some variable named `download` is missing its dollar sign. 

This is on Ubuntu 16.04 with the 4.8.17-galliumos kernel. The whole output (sans download progress bar) is this:

```
$ sudo apt install openrocket
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  openrocket
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 8,884 B of archives.
After this operation, 98.3 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 openrocket all 15.03 [8,884 B]
Fetched 8,884 B in 0s (17.6 kB/s)
Selecting previously unselected package openrocket.
(Reading database ... 226162 files and directories currently installed.)
Preparing to unpack .../openrocket_15.03_all.deb ...
Unpacking openrocket (15.03) ...
Processing triggers for man-db (2.7.5-1) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu5.1) ...
Processing triggers for gnome-menus (3.13.3-6ubuntu3.1) ...
Processing triggers for mime-support (3.59ubuntu1) ...
Setting up openrocket (15.03) ...
options :  --install --fast --verbose --
temporary directory: /tmp/openrocket.y0GIgcgNop
selected action = --install
installed version =
upstream version = release-15.03
wgetoptions= -nd -P .   -v --progress=dot:default  -t 3 -T 15
downloading navbar_download_button
https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03.jar...
--2017-12-12 23:51:55--  http://navbar_download_button/
Resolving navbar_download_button (navbar_download_button)... 198.105.244.114, 198.105.254.114
Connecting to navbar_download_button (navbar_download_button)|198.105.244.114|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘./index.html’

     0K                                                        20.9M=0s

2017-12-12 23:51:55 (20.9 MB/s) - ‘./index.html’ saved [291]

--2017-12-12 23:51:55--  https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03.jar
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/6010939/1804ffde-8205-11e5-8d23-744777cf4f23?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20171213%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20171213T075156Z&X-Amz-Expires=300&X-Amz-Signature=50b5d324bc010977a032e20b588f94d73080801cc681fc292a50397e41b65167&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DOpenRocket-15.03.jar&response-content-type=application%2Foctet-stream [following]
--2017-12-12 23:51:56--  https://github-production-release-asset-2e65be.s3.amazonaws.com/6010939/1804ffde-8205-11e5-8d23-744777cf4f23?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20171213%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20171213T075156Z&X-Amz-Expires=300&X-Amz-Signature=50b5d324bc010977a032e20b588f94d73080801cc681fc292a50397e41b65167&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DOpenRocket-15.03.jar&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.165.123
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.165.123|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 22211882 (21M) [application/octet-stream]
Saving to: ‘./OpenRocket-15.03.jar’

<<<<<<< DOWNLOAD PROGRESS BAR >>>>>>>

2017-12-12 23:52:42 (471 KB/s) - ‘./OpenRocket-15.03.jar’ saved [22211882/22211882]

FINISHED --2017-12-12 23:52:42--
Total wall clock time: 48s
Downloaded: 2 files, 21M in 46s (471 KB/s)
removing old .jar files from /usr/lib/openrocket ...
moving download to /usr/lib/openrocket ...
mv: cannot stat 'download': No such file or directory
```

Sorry if this is the wrong place to post this. If someone can point me in the right direction, I'd be happy to take it there.

teyrana, 2017-12-30:
Hmmm, `apt show openrocket` returns the following for me: (Ubuntu 17.04) 

Package: openrocket
Version: 15.03
Priority: optional
Section: universe/science
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Bdale Garbee <bdale@gag.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 98.3 kB
Depends: wget, default-jre
Homepage: http://openrocket.sourceforge.net/
Download-Size: 8,884 B
APT-Sources: http://us.archive.ubuntu.com/ubuntu zesty/universe amd64 Packages
Description: Model Rocket Simulator
 OpenRocket is a free, fully featured model rocket simulator that allows
 you to design and simulate your rockets before actually building and
 flying them.
 .
 OpenRocket features a full six-degree-of-freedom simulation, realistic
 wind modeling, a multitude of different components including free-form
 and canted fins, clustering and staging.
 .
 This package therefore downloads the pre-built OpenRocket .jar file from
 the upstream site and installs it.

wcravens, 2018-04-23:
> $ sudo update-openrocket --install --verbose 2>&1 | egrep -ai
> "erro|fail|warn"
> Resolving navbar_download_button (navbar_download_button)... failed: Name
> or service not known.
> ERROR: wget failed to download
>
> $ strings /usr/sbin/update-openrocket | egrep -ai "wget"
> wgetquiet=' -q '
> wgetfast='-t 3 -T 15 '
> wgetalways=' -nd -P . '
> wgetprogress=' -v --progress=dot:default '
> url=wget -qO - http://openrocket.sourceforge.net | grep https | grep
> download | awk -F\" '{ print $4 }'
> wgetoptions="$wgetquiet $wgetalways"
> [ "$verbose" != "yes" ] || wgetoptions="$wgetalways $wgetprogress"
> [ "$fast" != "yes" ] || wgetoptions="$wgetoptions $wgetfast"
> [ "$verbose" != "yes" ] || echo "wgetoptions=$wgetoptions"
> wget $wgetoptions $url
> || die_hard_with_a_cleanup "wget failed to download $downloadurl"
>
> So what's needed is to update the following snippet around line 130 or so:
>
> get_download_url() {
> url=wget -qO - http://openrocket.sourceforge.net | grep https | grep
> download | awk -F\" '{ print $4 }'
> }
>
> BUT! In practice it probably makes more sense not to scrape anything at
> all and just use git directly.

Absolutely.  The direct link is
https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03.jar

Do we need a link for the 'latest version' or would the pkg maintainer/s
update the pkg script on each release?  I vote for B.

Wes

dhbarr, 2018-04-24:
I'm in favor of the updater hitting a fixed URL for $latest which appropriately redirects to actual latest.

Took a shot at mangling the updater and line 172 appears to be strange nonsense in that "download" seems to be used but never defined?  Not sure if or how that ever worked.

teyrana, 2018-04-24:
Update: I had reached out to the current maintainer, "Bdale".   He didnt say alot, but it sounded like hed be open to updatung things if we know what the fix is.

wcravens, 2018-04-24:
Ah cool!  I had no idea that it was Bdale who maintained the package.  Totally makes sense though.

For everyone listening, if you don’t already know about Bdale, google Bdale Garbee right now! 🙂

And go buy an Altus Metrum something or other to atone for your sins.  Or build one, they are all open source!

Wes

bdale, 2021-02-26:
FYI, I'm current re-packaging OpenRocket from source for Debian, using the head of the unstable branch.   This will obsolete the current installer package, which has proven brittle, and which will be utterly worthless once bullseye is released due to removal of the ancient version of openjdk needed to make 15.03 actually run.  I'll try to remember to close this issue once an openrocket package makes it into Debian main unstable.

JoePfeiffer, 2022-11-21:
This is a report against the Debian package, not OR itself.  Also, the package maintainer says above it's being completely repackaged.

closed 2022-11-21 by JoePfeiffer