openrocket/openrocket #383: Transparency Issues in 3D views

teyrana, 2018-01-06:
Transparent (i.e. windowed) components show up as completely transparent in 3D views
-(Applies to both Photo Studio, and main window)

ChrisMickelson, 2018-01-31:
I believe I have a fix for this in my 383fix branch, it's based on your rc4 branch. Rather small change but it seems to work perfectly. 
https://github.com/ChrisMickelson/openrocket/tree/383fix

teyrana, 2018-01-31:
Nice!  

Are you comfortable rebasing and submitting the PR?  Or are you more looking for comments?

ChrisMickelson, 2018-01-31:
I'll try to rebase and submit a pull request. I'm still new to working with git but I'll give it a shot.

teyrana, 2018-02-01:
So before you go through all that, it looks like a pretty big PR....

for instance, the first commit in the series, dd25706f6c071aab954bf91901a83d49e3f4b369 shows 1 addition and 1,219 deletions.  Is this intentional?  Could you explain what all these changes are? 

and the biggest commit is probably with 7c5eba2745e6 with 3,183 additions and 2,795 deletions. 

this kinda looks like it was based off something old, and the merge ended up *enormous*.     What was the pr/23 based off?

ChrisMickelson, 2018-02-01:
Should be only 9 additions and 2 deletions in only one file. Not sure what's going on there. I was trying out Github Desktop for the first time as well. I'll remake the small changes to a fresh up-to-date branch later tonight.

ChrisMickelson, 2018-02-02:
My branches were way behind but now that I have a current one I submitted the pull request (#393) for this small fix.

wcravens, 2018-02-02:
Awesome.  This has been merged.

We’d

closed 2018-02-03 by teyrana