openrocket/openrocket #385: Build 18.01-rc4: cannot click-select components in 3d finished view in Windows

neilweinstock, 2018-01-18:
On this build I cannot click on components in the 3D view to select them.  This has never worked on the Mac (at least not since I've been using it), but it has always worked on Windows.  Now clicking on the 3D view in Windows has no effect.

neilweinstock, 2018-06-06:
I'm going to amend this one.  Working on Alpha5, click selection is simply scrambled and/or randomized.  See attached:
[Iris-T_home.zip](https://github.com/openrocket/openrocket/files/2078093/Iris-T_home.zip)

With the 3D view oriented with the nose on the left:

1. Clicking on the main BT generally selects the main BT
2. Clicking on different parts of the nose will select an assortment of other parts of the rocket.  Try it to see. 
3. Clicking on the rear portion of the rocket never selects anything.
3. Changing the orientation of the model in the 3D view can alter the behavior just a little.

neilweinstock, 2020-05-01:
Confirming that this problem is still present in latest builds.

teyrana, 2020-07-18:
From looking at RocketRenderer.java:68 ...

This is a more general issue, and applies to all 3d views on all platforms.  Basically, selecting instanced / booster / pods in 3d views probably doesn't work; it looks like it needs to be re-written.]

teyrana, 2020-07-19:
Addressed in #703

commit 0f2edc00 (2020-07-21): Merge pull request #703 from teyrana/385/3d-select

[fixes #385] Restores ability to select components in the 3d views
closed 2020-07-21 by teyrana