openrocket/openrocket #387: Build 18.01-rc4: CP calculation is bonkers

neilweinstock, 2018-01-18:
CP calculation is badly broken.  I'm not yet sure what triggered it.  See attached design; OR is showing me CP = -181 inches, which I'm pretty sure is not correct. :)

[plasma_dart2_scoop_newOR.zip](https://github.com/openrocket/openrocket/files/1644761/plasma_dart2_scoop_newOR.zip)

teyrana, 2018-01-27:
@neilweinstock I took a look at your rocket, and there are two backward facing transitions.  I'm not familiar with this particular model; are both of those supposed to be there?   

I ask because Openrocket, correctly calculates the CP on this rocket, *according to the Barrowman Equations*.

neilweinstock, 2018-01-29:
Interesting.  I’ll do some more experimentation and see what I can figure out, if anything.

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

neilweinstock, 2018-08-21:
I'm seeing this again and I can show some examples.  First, here's a simple test with pods:
[image]
This shows the CP way up front, despite the fact that it should clearly be towards the back, and in the text display in the upper right it says "N/A".  Note that in side view, no CP is shown, which perhaps corresponds more closely to "N/A", but I'm not sure why it should be N/A in the first place.

If I add a nose cone, CP moves way to the front.
[image]

In other cases, I'm seeing the CP being rendered off-axis _only in the 3D view_, and only when pods are involved, and when conditions are correct (I haven't figured out exactly what those conditions are).  Here's an example:
[image]
Note how the CP is shown axially centered on one of the pods.

Here's the ORK file used for the above screenshots.  If you delete the fins on the pods you can get the CP to jump all the way into the nose cone.

[cp_position.zip](https://github.com/openrocket/openrocket/files/2307796/cp_position.zip)

neilweinstock, 2018-09-11:
Here's another example of a problem using the latest alpha that Daniel sent me:
[image]

I put this file together for another issue, but again the CP is miscalculated to be at the front of the rocket instead of closer to the back.  It also is shown in the 3D view seemingly off axis in the wrong direction.  In side view this rocket shows CP as "N/A" for some reason.

[fin_decal_problem.zip](https://github.com/openrocket/openrocket/files/2372346/fin_decal_problem.zip)

wcravens, 2018-09-11:
Does it look like these would all be accurate if they were reversed, either
along the length of the rocket or 180 degrees around the CG?  I have to
wonder if there is a negative number multiplication error somewhere.

Wes

neilweinstock, 2018-09-11:
I tend to doubt it but really I’m just guessing.  There seem to be two separate problems here:

  1.  CP calculation: In this example, CP calculation is apparently failing, given that its position is reported as N/A.  I’m not sure how flipping any signs would make it better, but who knows what might be going on inside the algorithm.  In any case I don’t see why a design like this wouldn’t have a valid CP.  And note that this particular design doesn’t have any pods.
  2.  CP rendering in 3D view: Usually when pods are present in the design, CP is shown off-axis.  In the example I just posted, it’s possible that the CP shown in the 3D view is in fact complete nonsense and we should just ignore it, because if the program things CP position is N/A then who knows what numbers it’s actually using to calculate coordinates of the CP in 3-space.  Other designs (such as shown in previous comment) which have valid CP *and* incorrect display in 3D view are probably more appropriate for diagnosing this problem.  For this design I think the fact that it’s showing a CP at all is the problem.

  *   Neil

From: Wes Cravens <notifications@github.com>
Sent: Tuesday, September 11, 2018 3:26 PM
To: openrocket/openrocket <openrocket@noreply.github.com>
Cc: Neil WEINSTOCK <neil.weinstock@raritan.com>; Mention <mention@noreply.github.com>
Subject: Re: [openrocket/openrocket] Build 18.01-rc4: CP calculation is bonkers (#387)

Does it look like these would all be accurate if they were reversed, either
along the length of the rocket or 180 degrees around the CG? I have to
wonder if there is a negative number multiplication error somewhere.

Wes

________________________________

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

neilweinstock, 2018-09-11:
Here's the same file opened in 15.03:
[image]

Interestingly, the CP once again is calculated as N/A, and yet is displayed in the 3D view.  Although this time it is shown closer to what I would expect to be "correct" in the 3D view.  But I don't know for sure.  So I don't know if my example file actually shows a new calculation problem, maybe just the 3D view display problem.

Also: the file I attached above might have 3 fins rather than 1, due to a previous but that has since been fixed.  Setting number of fins to 1 creates the CP problem.  When #fins=3, the CP is calculated correctly (I think) but once again shown off-axis in the 3D view.

neilweinstock, 2018-09-11:
Sorry, seemed to have hit "close" by accident.

neilweinstock, 2018-09-24:
Here's a design that creates a similar bug in rendering the CG in 3D finished mode.
[image]

[cg_render_bug.ork.zip](https://github.com/openrocket/openrocket/files/2409427/cg_render_bug.ork.zip)

teyrana, 2018-09-30:
That last design: it's neutrally stable: there is no Center of Pressure for us to calculate.  I'm not sure what the expected output is, but I can't think of a better representation.

teyrana, 2018-09-30:
If we're talking about the example above, with the three detached pods, the 3D CoP seems to be displayed incorrectly in 3D (it's clamped to the x-axis in 2D).  I haven't calculated if the x value makes sense for that design, though.  That's probably the problem.

neilweinstock, 2018-10-01:
Both the CG and CP are sometimes shown off-axis, as shown above.

As for the design having no actual CP: if that's the case, then as far as I'm concerned nothing should be shown in the drawing.  The red dot indicates a CP.

teyrana, 2018-10-01:
> Both the CG and CP are sometimes shown off-axis, as shown above.
>
> As for the design having no actual CP: if that's the case, then as far as
> I'm concerned nothing should be shown in the drawing. The red dot indicates
> a CP.

Oh, I agree!  Ypure farther ahead than me- I was trying to ascertain if
there was an underlying calculation issue, or "only" a graphical /display
issue.

neilweinstock, 2018-10-01:
Fair enough! Good luck with your search.

ChrisMickelson, 2018-11-05:
Looking into this. So far it seems the 3D renderer is being told that CG and CP have a Y value, when any balanced design should be 0.
For example, here is some debug output I inserted into the renderer(RocketFigure3d.java:346) at the point where it's drawing the cg & cp carets:
CP: x=0.38327915090816433, y=0.008364311125459592, z=0.0
I can manually reset the Y value to 0 just prior to this point and it will *look* fine, but doesn't fix where the calculation is happening.
The offset CP seems to follow the first instanced fin attached to the main body tube.
The CG also seems to be pushed ~1% off-axis by pods, which is strange.

This issue is just as present with a simple 3FNC design.
[image]

teyrana, 2018-11-05:
Oh, I think I know whats going on.  (this is probably related to that last
fins PR).

What's the call site where its asking for the CP?

ChrisMickelson, 2018-11-05:
I started looking in where the cg & cp in the 3d view gets drawn
...\swing\src\net\sf\openrocket\gui\figure3d\RocketFigure3d.java in the method drawCarets()
around line 344 is where it's taking the cp and projecting them as 2d points before drawing the red dot. 
Working backward, RocketFigure3d.java get's it's local cp value set from RocketPanel:607, which comes from line 586/588, which comes from BarrowmanCalculator:61 which bounces around to line 205. 

That's as far as I can get for now.

That screenshot was just one of the built in examples.

neilweinstock, 2018-12-11:
Still not fixed in Alpha 10.  On the same design I referenced previously, the CP has gotten closer to the center axis, but still not quite there.

[image]

I'm having a little more trouble getting this condition to show up on other designs, so it has clearly gotten better.

neilweinstock, 2018-12-24:
Here's a version where I stripped everything out that doesn't affect the CP offset.
[alcubierre_cp.ork.zip](https://github.com/openrocket/openrocket/files/2707695/alcubierre_cp.ork.zip)

neilweinstock, 2019-02-05:
Alpha 11 report: 
CP seems properly on-axis in most of my tests, but still not this one:
[image]

I notice that in this one, the N/A CP is not shown in side view, but *is* shown in nonsense position in 3D view (this may or may not be a change from previous builds; just putting it here because I noticed it and we've discussed this design before).

Side view, no CP:
[image]

3D view, nonsense CP:
[image]

neilweinstock, 2019-02-07:
OK, on the *real* alpha 11, Alcubierre CP is shown properly on-axis in the 3D view.

I'm still seeing an off-axis CG problem, though, should I open a separate issue or address it here?  It's the cg_render_bug.ork file from several messages up:
[image]

neilweinstock, 2020-05-01:
I believe the CP problem has been fixed (haven't seen crazy off-axis CP in quite a while), but the off-axis CG problem is still there, so I'm going to close this and file another just for that problem.

commit 51418e34 (2018-12-08): [fixes #387] Fixes one source of off-axis CP error
commit 5e3d8013 (2018-12-08): Merge pull request #492 from teyrana/fix_387_offaxis_cp

[fixes #387] Fixes one source of off-axis CP error
commit eed58637 (2019-01-26): [fixes #387] Increases accuracy of Center-Of-Pressure calculations on instanced assemblies
closed 2020-05-01 by neilweinstock