openrocket/openrocket #388: Unable to run new install of OpenRocket

ndolnack, 2018-01-19:
First time install of Open Rocket (OpenRocket-15.03.jar) on IOS El Capitan 10.11.6 with Java 9.0.1+11 
I double click the .jar file to run it and the open rocket loading preview displays and shortly after I get an error stating "Open Rocket encountered an uncaught exception. This typically signifies a bug in the software.       _Runtime Exception:Error starting OpenRocket_ "

Was going to paste the bug report below but it wont allow me to copy it, I also attempted to send a bug report to the developer in the bug report section and get another error stating 
"[BugReportDialog.bugreport.dlg.failedmsg1]       OIException: Server responded with code 404, expecting 202"

Any help appreciated

teyrana, 2018-01-19:
Hi, this is a known issue.  We're working on a fix.

See #346 #347 for more detail

neilweinstock, 2018-05-25:
@ndolnack 
Please try this: http://www.rocketryforum.com/showthread.php?143540-OpenRocket-packaged-installers-for-Windows-and-Mac-to-solve-all-your-Java-problems

teyrana, 2018-05-25:
Openrocket 15.03 and Java 9 js a known incompatibility.  We're working on a
fix, and hope to have it in the next release.

muhamed-genco, 2019-03-10:
I have the same problem... How can we do ??

neilweinstock, 2019-03-28:
Did anyone try the link I gave above?

itshanney, 2020-04-06:
This should all be resolved now with latest builds after upgrading to Java 9. For 15.03 release, Java 8 is still required and will not work with Java 9+. Closing for now and please re-open if there's a fresh occurrence.

closed 2020-04-06 by itshanney

BrandonGThomson, 2020-04-09:
I am having the same problem. I have Ubuntu 18.04 with Java 11.0.6.  
Are there any known issues with Java 11 and openRocket?

JoePfeiffer, 2020-04-09:
> I am having the same problem. I have Ubuntu 18.04 with Java 11.0.6.
> Are there any known issues with Java 11 and openRocket?

As stated above, "For 15.03 release, Java 8 is still required and will not work with Java 9+." That includes 11.  You should be able to run the appimage at https://www.dropbox.com/s/25we2cz0ukc4bor/OpenRocket.AppImage?dl=0

BrandonGThomson, 2020-04-09:
@JoePfeiffer 
Thanks for the link to the software. Worked like a charm!