openrocket/openrocket #392: Random bug

phtdacosta, 2018-01-29:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.RuntimeException: Error starting OpenRocket
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40)
	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:32)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38)
	... 1 more
Caused by: java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(Unknown Source)
	at java.desktop/java.awt.EventQueue.invokeAndWait(Unknown Source)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(Unknown Source)
	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:76)
	... 6 more
Caused by: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.ConstructorInjectorStore.get(ConstructorInjectorStore.java:49)
	at com.google.inject.internal.ConstructorBindingImpl.initialize(ConstructorBindingImpl.java:125)
	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:521)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
	at com.google.inject.internal.FactoryProxy.notify(FactoryProxy.java:46)
	at com.google.inject.internal.ProcessedBindingData.runCreationListeners(ProcessedBindingData.java:50)
	at com.google.inject.internal.InternalInjectorCreator.initializeStatically(InternalInjectorCreator.java:133)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:106)
	at com.google.inject.Guice.createInjector(Guice.java:95)
	at com.google.inject.Guice.createInjector(Guice.java:72)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at net.sf.openrocket.startup.SwingStartup.runInEDT(SwingStartup.java:143)
	at net.sf.openrocket.startup.SwingStartup.access$000(SwingStartup.java:43)
	at net.sf.openrocket.startup.SwingStartup$1.run(SwingStartup.java:79)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue.access$500(Unknown Source)
	at java.desktop/java.awt.EventQueue$3.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$3.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
Caused by: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.MembersInjectorStore.get(MembersInjectorStore.java:65)
	at com.google.inject.internal.ConstructorInjectorStore.createConstructor(ConstructorInjectorStore.java:73)
	at com.google.inject.internal.ConstructorInjectorStore.access$000(ConstructorInjectorStore.java:28)
	at com.google.inject.internal.ConstructorInjectorStore$1.create(ConstructorInjectorStore.java:36)
[... 238 lines cut ...]
  88      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
  89      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.SingleFieldInjector.<init>(SingleFieldInjector.java:41)
  90      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.getInjectors(MembersInjectorStore.java:125)
  91      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.createWithListeners(MembersInjectorStore.java:95)
  92      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.access$000(MembersInjectorStore.java:34)
  93      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:42)
  94      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:39)
  95      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.FailableCache$1.apply(FailableCache.java:39)
  96      1.844 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:549)
  97      1.844 STDERR (SwingExceptionHandler.java:49) 	... 45 more
  98      1.844 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.ClassNotFoundException: javax.script.ScriptEngineFactory
  99      1.844 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.net.URLClassLoader.findClass(Unknown Source)
 100      1.844 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
 101      1.844 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
 102      1.844 STDERR (SwingExceptionHandler.java:49) 	... 67 more
 103      1.844 INFO   (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
 104      1.844 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 105      5.034 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 106      5.034 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

Windows 8.1
java version "9.0.4"
Java(TM) SE Runtime Environment (build 9.0.4+11)
Java HotSpot(TM) 64-Bit Server VM (build 9.0.4+11, mixed mode)

phtdacosta, 2018-05-25:
Thanks Daniel! Although I think it would be a lot better to revamp the code straight to Java 10. Oracle is going crazy and I know how hard it can be to maintain such an amazinh project really up to date with the language, but it seems that Oracle will be hopping thru new versions each time faster, and maybe your efforts to support Java 9 are going to be almost the same if you jump directly into the tenth version.

_Really thanks for this project, as a student and amateur rocket scientist this is helping me a lot!_

teyrana, 2018-05-25:
@phtdacosta 
Yea, that's a fair point.  We'd love for it to work with 10, as well! 

(Easier said than done, but sure!)

neilweinstock, 2018-05-25:
@phtdacosta 
In the meantime may I humbly suggest you check this out: http://www.rocketryforum.com/showthread.php?143540-OpenRocket-packaged-installers-for-Windows-and-Mac-to-solve-all-your-Java-problems

phtdacosta, 2018-05-25:
Amazing! I don't know who created this installer, and don't know if there is a dev involved with this project with enough Windows background, but it would be a great thing if, after each major OR update, the installers for windows/mac/ubuntu (.deb) get available "officially" for download and highlighted in the README

I believe in good will from the community members but getting an official installer from the main project repo would be great for general trust.

neilweinstock, 2018-05-25:
@phtdacosta 
That would be... wait a second, let me look that up... ah yes.  That would be me.  :) Glad it worked for you.

I have already suggested (#372) that future releases of OR should be distributed this way.  I think it's gonna happen.  Oracle is actually recommending it these days.

phtdacosta, 2018-06-04:
So thankful that the community around OR still breathing! As I said, I see how useful this software can be among many college students and serious hobbysts as there is virtually no other basic suite for rocket simulations. Just for curiosity matters, you guys know if major space corporations like SpaceX, ULA, Ariane and so on use some kind of commercial software or they develop their solutions by themselves?

teyrana, 2018-06-04:
Well basic is sort of the keyword there....  ig one had millions of dollars
yo throw ar this challenge,  the next tier of software is CFD computations
on a supercomputer.

Check out autodesks CFD. Module if you're curious:
 https://www.autodesk.com/products/cfd/overview

closed 2018-06-05 by phtdacosta