openrocket/openrocket #397: Please reopen issue 322

JoePfeiffer, 2018-03-27:
The problem with component analysis still exists.  It appears as if the contents of the elements of cgData is different in some circumstances than others; in some cases it's a coordinate (maybe coordinates of CG?
  not sure) and in others it's a mass (haven't worked out when each happens).  The exception occurs when a coordinate has been put in the array but a mass is expected.

teyrana, 2018-03-31:
Let's just use this issue.  What is the case you'd like to run a component analysis with? (or which GUI window / tab are you running it from?

teyrana, 2018-03-31:
This actually deserves more discussion.  I had noticed this earlier, but didn't have time then... 

Take, for instance,  `net.sf.openrocket.masscalc.MassCalculator#getCGAnalysis(...)@130`:

It currently returns a map of RocketComponent => Coordinate (of CG)...
This worked fine for simple object hierarchies, but once we implemented instanced objects, one Component may correspond to multiple locations and CGs.  

If we have a rocket with doubly instanced components, like so: 
```
- rocket 
  - nose
  - body
    - boosters (x2)
      - booster nose
      - booster body
        - booster fins (x3)
```

What, logically, should that analysis output? 
1. Should we generate an entry for each booster instance?  
2. Or an aggregated analysis of the average cg? 
   (including the total, centers of mass of all the boosters combined, and the aggregate total and center of mass for the fins.)

JoePfeiffer, 2018-03-31:
It occurs when, in the Rocket design window, you click Tools->Component Analysis and a motor is brought into the Stability table.  I'm seeing it both in RC4 and in the Unstable branch in the repository.

I've been looking at this since seeing the Exception, and I'd sort of like to disable the dialog for the next release.  I don't see it as terribly crucial, and it's a mess.  I've resolved the exception, and it turns out the motor CG relative to the motor mount, not the rocket (I'm looking at how best to resolve this at the moment).  I'm not seeing any effect from the sliders on any of the values in any of the tables (though they are affecting the values in the main view window).  Now that I've resolved the exception in my local copy, the total CG and Mass is 0.

As for what the analysis should output, I'd lean toward your (2), but clarifying much as you did in your example:  something like a column for the mass of a booster, and a second column giving the aggregate mass.  For that matter, the current table appears to be listing the components in the order given by a bread-first search of the tree representing the design; a more tree-like view would seem better.

If there are no objections, I'll keep looking at it...

teyrana, 2018-04-02:
I took a look at the mass calculator code, and im short, the back end needs a rewrite.

1) I'm okay with disabling that dialog
2) Also, if we can decide what it should look like, it wouldn't be too hard to fix.

teyrana, 2018-04-08:
Example Error as described above.
Occurs when simply trying to resize a vanilla 'Component Analysis' dialog. :(

```
---------- Exception stack trace ----------
java.lang.ClassCastException: java.lang.Double cannot be cast to net.sf.openrocket.util.Coordinate
	at net.sf.openrocket.gui.dialogs.ComponentAnalysisDialog$4.getValueAt(ComponentAnalysisDialog.java:215)
	at net.sf.openrocket.gui.adaptors.ColumnTableModel.getValueAt(ColumnTableModel.java:58)
	at javax.swing.JTable.getValueAt(JTable.java:2717)
	at javax.swing.JTable.prepareRenderer(JTable.java:5706)
	at javax.swing.plaf.synth.SynthTableUI.paintCell(SynthTableUI.java:683)
	at javax.swing.plaf.synth.SynthTableUI.paintCells(SynthTableUI.java:580)
	at javax.swing.plaf.synth.SynthTableUI.paint(SynthTableUI.java:364)
	at javax.swing.plaf.synth.SynthTableUI.update(SynthTableUI.java:275)
	at javax.swing.JComponent.paintComponent(JComponent.java:780)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JViewport.paint(JViewport.java:728)
...
```

commit 0b1e0113 (2018-04-08): [Resolves #397][Workaround] Hides the Component Analysis Dialog until it can be fixed.
closed 2018-12-02 by JoePfeiffer