openrocket/openrocket #401: Ability to hide components in 3d views

snrkl, 2018-05-08:
It would be particularly helpful if it was possible to hide individual objects in the 3d rendering, such that you could see the elements that it might be covering.

I find this is needed when fine tuning measurements for elements such as fins with tabs, internal tubes on more complex fiddly rockets and when trying to take screenshots of internal components where bulkheads, centering rings or nosecone shoulders are in the way.

neilweinstock, 2018-05-25:
This would be useful as proposed.

I would take it a step (or two) further and say that it would be *extremely* helpful to be able to select, for every component, whether it is (a) displayed, and (b) simulated.  Default (current behavior) is both selected.  But oftentimes I'll have items that are in there for visual purposes and I don't want it to try to figure out how to simulate them, or other things (e.g., base drag compensators) that are there for simulation but I don't want to see in my visualization.

hcraigmiller, 2021-06-23:
Now that the transparency setting has been fixed, would simply setting the component to 100% transparency (effectively hiding it in 3D views) be sufficient to close this issue?

neilweinstock, 2021-06-23:
For now, I would say yes.  It’s a hack but it works.

It’ll be an even better hack when the transparency is set in the appearance tab rather than the color picker.

hcraigmiller, 2022-01-24:
In reviewing the context of the comments, it appears that the desired outcome is invisibility for user selected components in 3D views. Where the issue creator did not describe how the desired change might be accomplished (check box, slider, or other), and in light of the comment of @neilweinstock above, I would suggest three options:

1. That the transparency slider be located on BOTH the _Appearance_ tab, in the _Appearance_ section, inserted between _Color_ and _Shine_, AND as currently exists on the _HSV_ and _HSL_ color picker pages (in any event, the transparency slider needs to be added to the _RGB_ and _CMYK_ color picker pages). This change will allow the user to test the level of transparency on the color picker page, and to allow adjustment of the transparency on the appearance tab without having to go any deeper into the color page option settings. This should be a fairly quick fix,
2. If grouping is desired, a selection check box (labeled "Use group transparency") could be placed on the component _Appearance_ tab, in the _Appearance_ section, inserted below the _Use default_ check box, with an activation button (labeled "Transparency") being placed on the _3D Unfinished_ and _3D Finished_ views, below the _Delete_ button. When this button is pushed, all selected components become 100% transparent. When pushed again, the selected components return to the transparency set on the component's appearance and/or color picker pages (in any event, the transparency slider needs to be added to the _RGB_ and _CMYK_ color picker pages). This would be more of a new feature than a quick fix.
3. Do both; the best of both worlds. The first could be done sooner, with the latter being more of a new feature request.

I would lean toward 3.

neilweinstock, 2022-01-25:
I have been meaning to submit a feature request for exactly (1).  It should be very easy, and it is the Right Way To Do It.  No mods are required to the color picker, all that's needed is one new control.  Opacity is sort of an orthogonal parameter to color, much like shine, and right now it is too far buried in the color picker.  Of course, it is also not likely to be used that often, but I think it's justified to go into the appearance tab.

Although (1) would accomplish what @snrkl was asking for, it's really not exactly the same thing.  I believe that I will still propose (1) as a separate feature request.

It is very possible that fancier functionality could be useful in the future, like (2), which is a bit trickier and should be kept as a separate feature request (or maybe just keep it here).  Likewise, there is still a possible notion of a checkbox to just completely hide the object, so it would be easy to toggle on and off.  It would then require *another* checkbox to implement (2) (hide children).

It would be further possible to implement show/hide toggle in the tree view, which is where I think it actually makes the most sense.  But that's really far afield.

And so I conclude: split off (1) as a separate feature (I will do that), and leave this one open as a request to do something a little more directed at the original proposed use case.

neilweinstock, 2022-06-10:
I feel like this one should be closed, since the opacity slider has been implemented and works well.  This seems to adequately address the original feature request.

Thoughts?

hcraigmiller, 2022-06-10:
Whole heartedly agree.