openrocket/openrocket #409: Maven-ize the OR project

SpamWall, 2018-05-29:
I know this isn't a high priority, but we should convert the project to use Maven.

wcravens, 2018-05-29:
Just for posterity would you mind putting up some
pros/cons/advantages/risks/etc for the change.  Also, to give other people
a chance to discuss.

Thanks,

Wes

itshanney, 2020-03-19:
+1 vote for Mavenizing the project. The biggest pro for the migration is that Maven includes a dependency management system so we no longer have to manually include JAR files of dependencies. You would declare the dependency in your pom.xml file and Maven handles the dependencies for you, including when you package a JAR file, run tests or any other action that needs to define/set the classpath.

Another plus is that Eclipse, IntelliJ and every major IDE automagically supports a Maven project out of the box. If you import the Maven project, the IDE will automagically setup the project for you and it "just works". We will no longer have to maintain .classpath, .project, .iml or .idea files for organizing the project - just define a standard set of instructions for importing through IntelliJ or Eclipse.

The biggest con will be that the project folder structure will change dramatically. Maven relies on a set of conventions that define the folder hierarchy and there will be a change there that people will have to get used to. 

Overall, I would say the benefits are worth it and it unifies some things, like general project setup in IntelliJ/Eclipse along with dependency management, which are huge wins.

@wcravens @teyrana I'm happy to take this issue and produce a working setup in a branch for review.

teyrana, 2020-03-19:
I don't have much experience with maven specifically.  Offhand the big
concerns I have are:

Caveats:

   1. We will still want to pin the dependency versions.  (One assumes
   Maven supports this).
   2. Changing the folder structure is a *big* change:
      1. Some of us know the project structure, and it will impose a
      learning cost on existing developers.  It might be worth it, over time,
      though.
      2. This will give you an opportunity to rationalize some weird
      oddities of the existing folder structure.
      3. There are a lot of classes dynamically loaded by inspection.
       Expect a lot of cryptic method not found errors that only occur run-time.

Daniel Williams
equipoise@gmail.com
858-220-8134 (Mobile)
github.com/teyrana
<https://github.com/teyrana>LinkedIn
<https://www.linkedin.com/in/williamsdm>

thzero, 2020-05-19:
3. There are a lot of classes dynamically loaded by inspection.
       Expect a quite a few cryptic "method not found" errors that only occur run-time.

Not sure that is too much of a concern.  Gradle (or Maven) is a dependency management and build system.

I'm not sure why one would use Maven when Gradle is available.

I have a full gradle revamp of the build system, etc. done for some time on my fork.  Its horrible out of date.

https://github.com/thzero/openrocket/tree/gradle-unstable

Frankly I've been waiting for "unstable" to get promoted to master and a new release done (with locked down feature set) before submitting as big of change as a PR.
Think its far better to bundled up the current feature set, release it than introduce a huge change.

teyrana, 2020-05-24:
> 3.  There are a lot of classes dynamically loaded by inspection.
>       Expect a quite a few cryptic "method not found" errors that only occur run-time.

To be clear, I don't mean DI errors (although those DO exist in openrocket, I consider those a separate category) 

I mean things like: the UI will dynamically look up the "getOuterDiameter" method for BodyTubes.  But there's no string to grep of "getOuterDiameter", and you can't trace it through the IDE, because it's not a hard, "link-time" error.  It only happens when that code path is executed.

Certainly doable of course, but it will take some time.

> Frankly I've been waiting for "unstable" to get promoted to master and a new release done
> (with locked down feature set) before submitting as big of change as a PR.
> Think its far better to bundled up the current feature set, release it than introduce a huge change.

Yea, that strikes me like the right approach.

thzero, 2020-05-24:
> I mean things like: the UI will dynamically look up the "getOuterDiameter" method for BodyTubes. But there's no string to grep of "getOuterDiameter", and you can't trace it through the IDE, because it's not a hard, "link-time" error. It only happens when that code path is executed.

Yeah, should not have any impact with using either Maven or Gradle; they are compile only.

> Yea, that strikes me like the right approach.

Is the feature set locked down?

teyrana, 2020-05-25:
(a) the dynamic look-up things were in relation to the idea of reorganized the folder structure.
     If all you're doing is migrated the existing folder structure to Maven or Gradle.... yea, this is a much smaller changeset, and not a big deal.

(b) We are in a feature freeze, yes.  But still fixing bugs.

thzero, 2020-05-25:
> (b) We are in a feature freeze, yes. But still fixing bugs.

Thats good.

closed 2024-04-09 by SiboVG