openrocket/openrocket #418: Suggestion: show pod locations in side and back views

neilweinstock, 2018-06-06:
It would be nice to see the pod locations in the side and back views.  I'm not sure how they should be shown, but it is always useful to get a bit of visual feedback what you're doing.

My thinking is that they are not needed in the 3D views.

teyrana, 2018-06-06:
Please note: the pod's front/middle/back position is calculated based on
the pod's contents.

neilweinstock, 2018-06-07:
At any given time, though, there's a single point that represents the location of each pod, is there not?

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2018-06-07:
So, the positioning works like Fins on a bodyTube, (except more
complicated)

When positioning via 'TOP'  you supply the distance from the top (front) of
nthe body tube to the top of the fins.  And when using bottom, the distance
is from Bottom-to-Bottom.   So the point changes based on the positioning
method.

In addition:  the "front" of a pod or stage is specified by it's contents.
 So if you add a component to a 'TOP'-positioned stage, the bottom of  the
stage move.... if you add a component to a 'Bottom'-positioned pod, the top
will update/move.  (internally, the position is calculated as the
forward-most point of the assembly)

Does that answer your question?

Daniel Williams
equipoise@gmail.com
858-220-8134 (Mobile)
github.com/teyrana
<https://github.com/teyrana>LinkedIn
<https://www.linkedin.com/in/williamsdm>

neilweinstock, 2018-06-07:
I confess I didn’t really have a question.

My desire is to *see* the root position of the pod indicated on the side and/or rear views.  I know it can be in different places based on different settings. It would be helpful to see where it is.  It would then be doubly useful to be able to double click on it.

I had no expectation to get this feature in this release, so it’s probably not worth too much thought process right now.  Just wanted to register it for future consideration.

  *   Neil

equipoise@gmail.com<mailto:equipoise@gmail.com>
<https://www.linkedin.com/in/williamsdm>

neilweinstock, 2022-08-30:
This was fixed in 22.02 beta5.

closed 2022-08-30 by neilweinstock