openrocket/openrocket #419: Freeform fin editor improvements

neilweinstock, 2018-06-07:
I use the freeform editor a lot.  It works but has lots of pain points.  Here's a list.  I hope to be able to contribute some of these changes myself in the future.
1. Often, new points appear in the wrong place.  It's kind of maddening and it becomes a game to try to figure out where to click to make the new point appear on the correct segment.
2. When the window is enlarged, the layout does not respond in an intelligent way.  The typical desire would be for the actual editor area to be enlarged, but instead you get lots of expansion of other areas, and only a little enlargement of the editor itself.  See attachment.
[image]
3. The scrollbars in the editor area do not adjust when the zoom value is changed.  Therefore, if you zoom in, you often cannot scroll to access the entire fin area.  See attachment.
[image]
4. There is no correlation between the coordinates on the left and the vertices on the editor.  If I want to hand-edit the X/Y values for a particular point, it can be very difficult to figure out which line to edit.  When a row is selected in the coordinate editor, the corresponding vertex in the editor should be highlighted.
5.  I would be super-nice if there were a way to automatically align a point vertically or horizontally with adjacent points (e.g., holding down a modified while dragging or something like that.)

teyrana, 2018-07-07:
@neilweinstock can you elaborate on the new points thing? How are you creating new points? where do they appear?  where do you expect them to appear?

neilweinstock, 2018-07-07:
I’ll get some screen shots to illustrate.

I figured these free form editor issues would wait until next release; they are annoying but not critical.

- Neil

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2018-07-07:
I'll leave the Feature Requests alone, but the bugs are probably worth fixing.   (also, I've already fixed a few of them.)

there are a couple cases: 
[1] edit existing point by click + drag
[2] click on existing edge, to add a new point in between existing points, on that edge.
[3] double-click on point table?
[4] other clicks on blank space :  ??? 
     (IMHO, these clicks should just do nothing)

neilweinstock, 2018-07-09:
#2 is the one that vexes me.  Here are some attachments to illustrate.
First, the OR file I'm using:
[alc_new_pods.zip](https://github.com/openrocket/openrocket/files/2177124/alc_new_pods.zip)
Edit the "wings" component, and open up the shape editor.  Here's the problem:
[image]

neilweinstock, 2018-08-06:
I haven't been able to break it in Alpha 8.  Awesome!

neilweinstock, 2018-08-15:
I noticed one issue with the new implementation: the new point is always placed at the midpoint of the segment, rather than where the mouse was clicked.  This is not a tragedy, but it's not the behavior I would expect.

I would expect either:
a) new point is exactly where the mouse is clicked (as before), or
b) new point is on the segment, at the point closest to where the mouse was clicked.

teyrana, 2018-08-16:
Yea, that was the preexisting behavior.  I can update it, though.  I agree,
that makes more sense.

neilweinstock, 2018-12-11:
This one seems completely fixed these days.

commit eee24233 (2018-09-29): [fixes #419] Clicking in fin-point figure now calculates closest segment correctly
closed 2018-12-11 by neilweinstock