openrocket/openrocket #425: FinSetConfig Improvements (3/5)

teyrana, 2018-06-22:
shape_editor_zoom
The scrollbars in the editor area do not adjust when the zoom value is changed. Therefore, if you zoom in, you often cannot scroll to access the entire fin area.

neilweinstock, 2018-07-09:
This one looks much better now.

neilweinstock, 2018-08-06:
Scrollbars work correctly in Alpha 8.

neilweinstock, 2018-08-15:
In Alpha 8 the problem still exists, apparently.  Haven't been able to try whatever Joe added in his last commit.

Here's the problem:
[image]

Here's the ork file.
[fin_shape_scroll.zip](https://github.com/openrocket/openrocket/files/2291509/fin_shape_scroll.zip)

teyrana, 2018-08-19:
Yup. Confirmed.  That is pretty blatant.

commit 80c0fa85 (2018-09-29): [fixes #425] FinPointFigure ScrollBars now adjust with zoom in/out
closed 2018-10-21 by teyrana