openrocket/openrocket #426: FinSetConfig Improvements (4/5)

teyrana, 2018-06-22:
There is no correlation between the coordinates on the left and the vertices on the editor. If I want to hand-edit the X/Y values for a particular point, it can be very difficult to figure out which line to edit. When a row is selected in the coordinate editor, the corresponding vertex in the editor should be highlighted.

neilweinstock, 2018-08-06:
This is fantastic.

Might I suggest a slightly different way of highlighting the selected point?  Right now you have them growing slightly, but not centered on the same point.  It's subtle and a little weird.

How about filling the selected square with translucent gray or something like that?  E.g:
[image]

teyrana, 2018-08-07:
Haha, yea, I did that because it was easy.  I must admit I dint spend a loy
of time thinking about it.

neilweinstock, 2018-08-07:
It’s not a big deal; having the feature at all is more important than anything, but I figured a slightly more visible highlight would be easy enough to implement.  I also thought of changing the outline of the selected point to green (that would probably look nice but not good for the colorblind) or adding a second green outline outside the first.

Attached are mockups of a few additional different approaches.  Pick whichever you like, or a new one. 😊

[cid:image001.png@01D42E2D.5420D1B0]

  *   Neil

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2018-08-08:
Well, its not a big deal either way.  Y'all discuss, And decide what makes
most sense :)

commit 4a91ecd6 (2018-09-29): [feature][Resolves #426] implemented FinPoint SelectedIndex Indicators

- figure and table update each other
closed 2018-10-21 by teyrana