openrocket/openrocket #427: FinSetConfig Improvements (5/5)

teyrana, 2018-06-22:
Feature Config:

"I would be super-nice if there were a way to automatically align a point vertically or horizontally with adjacent points (e.g., holding down a modified while dragging or something like that.)"

Option 1:  Snap to nearest point / vertical / horizontal (when within 5? x? pixels? mm?)
Option 2: Snap to grid (with tolerance: 5? mm?)
Option 3: Manual Edit via point table

neilweinstock, 2018-08-13:
I like option 1, when used in combination with a modifier key.  I'm not sure of the details.  Would probably take some experimentation.

SiboVG, 2022-08-25:
Would also be nice to snap to points on the parent component, e.g. snap to the end point of the parent transition when holding down a modifier key.

neilweinstock, 2022-08-25:
I'll be working on snap-to-45 degree increments relative to previous point, while holding down shift key.  

Not sure about snapping to endpoint of parent, unless we employ yet another modifier key.  How would that work exactly?

SiboVG, 2022-08-25:
> Not sure about snapping to endpoint of parent, unless we employ yet another modifier key. How would that work exactly?

I would use the shift-key for snapping to 45-degree increments (you will also include vertical and horizontal positions, right? not only diagonal 45° directions?) and the meta-key (ctrl on Windows/Linux, Cmd on macOS) for snapping to the grid or the parent. You should add a check if both modifier keys are pressed, in which case you should give priority to one of the two snap methods.

neilweinstock, 2022-08-25:
Yes, 0 and 90 are valid 45-degree increments. :)

I'm not sold on the other functionality.  Will think about it and experiment once I get down to playing with the code.

SiboVG, 2022-08-25:
> I'm not sold on the other functionality.

You mean snapping to the grid or other points in general?

neilweinstock, 2022-08-25:
Either snapping to the grid or to endpoint of the parent component.

The grid is problematic because it auto-adjusts according to zoom level... exactly what grid are we snapping to?  Snap to parent edge could be a problem if you accidentally activate it when you're attaching to a body tube, and suddenly you've moved the fine point 20" to the left, changing zoom level and all of the coordinates of all the other points.

SiboVG, 2022-08-25:
> The grid is problematic because it auto-adjusts according to zoom level...

That's not really problematic code-wise, you can always query the current grid size/points.

> Snap to parent edge could be a problem if you accidentally activate it when you're attaching to a body tube, and suddenly you've moved the fine point 20" to the left, changing zoom level and all of the coordinates of all the other points.

I don't see what that has to do with the snapping? You have the same "issue" without snapping. The snapping is only valid while you hold the modifier key, after that, the fin point is fixed just like you would fix a fin point by just clicking and dragging in the fin editor.