openrocket/openrocket #431: Can't save fins on pods with fin count=1

neilweinstock, 2018-07-09:
This is on Daniel's build, still labeled alpha 5.
Using this design:
[alc_new_pods.zip](https://github.com/openrocket/openrocket/files/2177152/alc_new_pods.zip)
On the pods, there are two fins: "fin tip" and "pylon".  When opening the file, each of those will open with a fin count of "3", despite the fact that I save the file with fin count of "1".

neilweinstock, 2018-08-06:
Works correctly on the file that was previously failing in Alpha 8.

teyrana, 2018-08-08:
Sorry, can you rephrase that? alpha8 should have fixed it :/

neilweinstock, 2018-08-08:
Sorry, I knew the wording sounded funny when I wrote it.

It seems to work now.  I tried it on the file that was previously breaking, and it reads back correctly with one fin in each set.

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2018-08-08:
Understood.  Thanks for getting back to us :)

neilweinstock, 2018-08-17:
This problem has re-appeared for me in alpha 8.  See attached ORK file.  The trapezoidal fins on the pods should have 1 per set, but they always come back as 3 when I reload the file.
[faddle.zip](https://github.com/openrocket/openrocket/files/2298305/faddle.zip)

neilweinstock, 2018-11-03:
I still see this problem in Alpha 9.

neilweinstock, 2019-04-07:
I'm re-opening this one.  In the alpha I'm using now, pods with instance count = 1 are again not being saved.  Or at least, when re-opening the file they all show up with instance count = 2.

teyrana, 2020-04-25:
@neilweinstock 

This should be fixed in unstable, now.   Are you stil seeing this?

neilweinstock, 2020-04-27:
Nope, seems to be fixed.

neilweinstock, 2020-04-27:
Time to close it down again, hopefully for good this time. :)

commit 1418cb90 (2018-09-29): [fixes #431] Fins default to instance count / fin count == 1

- Fixed init bug
- added unittests for fin count loading/saving/creation
closed 2020-04-27 by neilweinstock