openrocket/openrocket #433: Cannot Launch OpenRocket 15.03 on new Mac running JRE 10.0.2

snrkl, 2018-08-01:
System:
Mac OSX High Sierra: 10.13.6 (17G65)
Freshly installed JRE 10.0.2

Error:

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.RuntimeException: Error starting OpenRocket
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40)
	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:32)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38)
	... 1 more
Caused by: java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(Unknown Source)
	at java.desktop/java.awt.EventQueue.invokeAndWait(Unknown Source)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(Unknown Source)
	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:76)
	... 6 more
Caused by: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.ConstructorInjectorStore.get(ConstructorInjectorStore.java:49)
	at com.google.inject.internal.ConstructorBindingImpl.initialize(ConstructorBindingImpl.java:125)
	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:521)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
	at com.google.inject.internal.FactoryProxy.notify(FactoryProxy.java:46)
	at com.google.inject.internal.ProcessedBindingData.runCreationListeners(ProcessedBindingData.java:50)
	at com.google.inject.internal.InternalInjectorCreator.initializeStatically(InternalInjectorCreator.java:133)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:106)
	at com.google.inject.Guice.createInjector(Guice.java:95)
	at com.google.inject.Guice.createInjector(Guice.java:72)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at net.sf.openrocket.startup.SwingStartup.runInEDT(SwingStartup.java:143)
	at net.sf.openrocket.startup.SwingStartup.access$000(SwingStartup.java:43)
	at net.sf.openrocket.startup.SwingStartup$1.run(SwingStartup.java:79)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue.access$600(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
Caused by: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
[... 256 lines cut ...]
  85      2.898 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:529)
  86      2.898 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
  87      2.898 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
  88      2.898 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
  89      2.898 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
  90      2.898 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
  91      2.898 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.SingleFieldInjector.<init>(SingleFieldInjector.java:41)
  92      2.898 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.getInjectors(MembersInjectorStore.java:125)
  93      2.899 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.createWithListeners(MembersInjectorStore.java:95)
  94      2.899 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.access$000(MembersInjectorStore.java:34)
  95      2.899 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:42)
  96      2.899 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:39)
  97      2.899 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.FailableCache$1.apply(FailableCache.java:39)
  98      2.899 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:549)
  99      2.899 STDERR (SwingExceptionHandler.java:49) 	... 45 more
 100      2.899 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.ClassNotFoundException: javax.script.ScriptEngineFactory
 101      2.899 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.net.URLClassLoader.findClass(Unknown Source)
 102      2.899 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
 103      2.899 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
 104      2.899 STDERR (SwingExceptionHandler.java:49) 	... 67 more
 105      2.899 INFO   (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
 106      2.900 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 107      9.330 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 108      9.337 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

kernelsmith, 2019-03-14:
yeah, this is kind of a dupe of #347. For some reason the max compatible version of 8 is not mentioned anywhere obvious so we all keep opening issues like this

itshanney, 2020-04-09:
Closing issue as this is a known issue with OpenRocket 15.03 not being able to run on versions of Java newer than Java 8. FAQ has been updated to address this issue and hopefully alert users in the future.

https://github.com/openrocket/openrocket/wiki/FAQ

closed 2020-04-09 by itshanney