openrocket/openrocket #454: Freeform fin editor: can't move the first root point (0,0)

neilweinstock, 2018-10-11:
In latest alphas that I've tried, it is impossible to move the first root point at 0,0.  This is a new breakage; in the previous release it is possible to drag the 0,0 point, which stays at 0,0 but causes all the other points to move accordingly.

teyrana, 2018-10-11:
Sonuva.

I specifically fixed that. :(   What broke?

neilweinstock, 2018-10-16:
Once the build is fixed I'd be happy to do some more testing if you'll shoot me a jar file.

teyrana, 2018-11-04:
@neilweinstock Nominate to close.

neilweinstock, 2018-11-05:
In alpha 9 this problem persists.

neilweinstock, 2019-02-05:
Looks generally fixed in Alpha 11.  Only remaining weirdness is that if you try to drag the root point to the left of the parent object, at that moment it'll create a new point and leave the root point at the left edge.

Without having really thought about it too much, I think it's probably OK to leave it this way for now.

neilweinstock, 2019-02-05:
Actually the remaining behavior described above might be related to #471.  Perhaps should leave this open for review until that one is closed.

teyrana, 2019-02-06:
hrm?  That was exactly the behavior this was supposed to fix.

Okay, I'll have to take another look at this, then.

neilweinstock, 2019-02-07:
Looks fixed in the *real* alpha 11.  Closing.

commit 22bda82a (2018-11-26): Merge pull request #485 from teyrana/fix_454_cannot_move_p0

[fixes #454] Allows dragging of first point
commit 54af8fca (2018-11-26): [fixes #454] Allows dragging of first point
closed 2019-02-07 by neilweinstock